WildernessLabs/Meadow_Issues #125: System.Net.HttpClient Hangs if Wi-Fi is not connected first

MarkAlanJones, 2021-01-02:
it seems
     if you create an httpClient instance - and forget to connect the meadow WiFi first, the network calls will hang and never timeout

`var result = await client.GetAsync(url);`

it would be nice if the low level meadow code could detect that 
- the WifiAdapter is not initialized
- the WifiAdapter is initialized but not connected to an access point

and throw an exception

or throw a timeout exception at least - but user will not be able to figure out what is wrong as easily.

NevynUK, 2021-03-22:
Suggesting that in this case the ESP32 responds with network down and generates .NET generates the corresponding exception:

`App: Unhandled Exception: System.AggregateException: One or more errors occurred. (Network subsystem is down) ---> System.Net.Http.HttpRequestException: Network subsystem is down ---> System.Net.Sockets.SocketException: Network subsystem is down`

NevynUK, 2021-07-06:
Fixed and will be present in the release following 0.5.1

closed 2021-08-05 by NevynUK