WildernessLabs/Meadow_Issues #126: Device.WiFiAdapter.Scan or Device.WiFiAdapter.Connect seems to hang often - but will eventually work if redeployed

MarkAlanJones, 2021-01-03:
I have been trying to add WiFi support to a new project and find it is hanging often
so i returned to the WiFi_Basics project that was published when WiFi was first enabled

beta 4.2 enabled scanning before connecting...

essentially:
`Device.InitWiFiAdapter().Wait();`
`Device.WiFiAdapter.Scan();`
`Device.WiFiAdapter.Connect(Secrets.WIFI_NAME, Secrets.WIFI_PASSWORD);`

I find it often hangs during the scan - normal scan is less than 10 seconds before results are returned
if the scan is successful the connect will often hang similarly

if it makes it all the way through it is successful and can get data from the internet

I can't tell what changes between the success and the Hang state .
I just keep redeploying - maybe clean and rebuild in between deploys

TImeout detection or other error state in the driver could be surfaced to the caller through an exception ?

miloush, 2021-03-04:
I am having the same problem, except that I haven't been able to connect yet.

Rarely the `Connect` returns `UsnpecifiedFailure`, mostly it just hangs. If I put `Scan` before `Connect`, the `Scan` hangs - haven't seen a finished scan yet. I tried redeploying, power cycling, nothing helps.

Am I missing some basic trick to get the WiFi working?

thegreatco, 2021-03-09:
Hi folks, if you're still running bet 4.2, please upgrade to beta 4.5 and let me know if this issue is still occurring.

upswing1, 2021-03-10:
I have the same issue with beta 4.5
I cannot get the WIFI to connect using the sample app
I have included a screenshot of the dependencies that the project is using.
The application hangs when calling the connect method:    var result = Device.WiFiAdapter.Connect(Secrets.WIFI_NAME, Secrets.WIFI_PASSWORD);
The wifi network has the following characteristics:
SSID:	Meadows
Protocol:	Wi-Fi 4 (802.11n)
Security type:	WPA2-Personal
Network band:	2.4 GHz
Link speed (Receive/Transmit):	144/144 (Mbps).
Following the instructions for windows I have included post processing instructions in the project file:
 <Target Name="PostBuild" AfterTargets="PostBuildEvent">
    <Exec Command="copy C:\Users\PC\.nuget\packages\wildernesslabs.meadow.assemblies\0.5.0\lib\net472\System.Net.Http.dll  $(OutDir)" />
  </Target>
meadows is running the following
MeadowOS Version: 0.4.5 (Feb  6 2021 03:00:34)
CoProcessor: ESP32, CoProcessor OS Version: 0.4.5.0, Mono Version: 0.0.0.1, Device Name: MeadowF7
Can anyone spot anything incorrect?  Thank you
[image]

upswing1, 2021-03-10:
In my case It never works even after many redeploys.

bryancostanich, 2021-03-10:
per our conversations internally, we're going to try and surface better diagnostic errors from the ESP32

NevynUK, 2021-03-22:
Beta 0.4.6.1 has better diagnostics returned when connecting.

Is this problem still occuring and if it is what information is being given in the status code?

NevynUK, 2021-06-08:
Closed as no activity for more than 1 month.

closed 2021-06-08 by NevynUK

jessetabali, 2022-05-02:
> Beta 0.4.6.1 has better diagnostics returned when connecting.
> 
> Is this problem still occuring and if it is what information is being given in the status code?

I am still having same hang issue with beta 6.3.