WildernessLabs/Meadow_Issues #131: Better app exit messaging

ctacke, 2021-01-18:
meadow apps generally are designed to never exit.  if you make a mistake, though, and it does (like forgetting a Sleep(Infinite) call), then the app quits and you get something like this out the Console:

```
App: * Assertion at object.c:585, condition `is_ok (error)' not met, function:mono_runtime_class_init_full, (null) assembly:/meadow0/mscorlib.dll type:ThreadAbortException member:(null)
```

A more useful/helpful message like "The meadow Application has Exited" would be nice to prevent me trying to find some other bug in my code.

bryancostanich, 2021-03-12:
I think we can solve this in the `IApp` re-arch work. we can tackle it along with the other lifecycle stuff we need to work on.

bryancostanich, 2022-07-03:
Relevant to App Lifecycle work.

bryancostanich, 2022-08-09:
@ctacke with the new lifecycle, this is fixed, yeah?

patridge, 2022-10-27:
Closing as resolved now that [RC-1 is released](http://developer.wildernesslabs.co/Meadow/Release_Notes/Release-Candidates/).

closed 2022-10-27 by patridge