WildernessLabs/Meadow_Issues #132: Can't Deploy app to meadow anymore.

RudyMeijer, 2021-01-20:
I've been able to deploy the Hello Word app to meadow succesfully via VS2019.

Next thing i tried was generating 1kHz on pin D15. This went also successfully.

[image]

I get a nice block signal of 1kHz on pin D15 on my oscilloscoop.
[image]

However: All further deployments result in an error.
[image]

And meadow.CLI command result also in an error.
[image]

Could the absence of a `Thread.Sleep(duration);` be the problem? (Meadow has no time to communicate with the CLI)

Things i did so far without result:
Power off/on PC and meadow.
Follow "Getting started" again. (Zadig com4, --FlashOS)

I use latest version of Meadow and VS2019.
[image]
[image]

Any suggestion is welcome.
Rudy Meijer

Dblm0, 2021-01-21:
@RudyMeijer, I faced the same problem several times. It seems like you're right about `Thread.Sleep` absence.
To fix this Meadow state I came up with the following workaround:
 - Downgrade `Meadow.OS` with `DFU` to prevent loading the App,
 - Erase memory with `Meadow --EraseFlash` CLI command,
 - Upgrade` Meadow.OS` to actual version.
 
 This method might not be optimal, but I don't know the other ways to perform that clearing process.

RudyMeijer, 2021-01-21:
@Dblm0, Thx, i will try your suggestion and let know the result. b.t.w. to which version you downgrade?

Dblm0, 2021-01-21:
> 
> 
> @Dblm0, Thx, i will try your suggestion and let know the result. b.t.w. to which version you downgrade?

0.4.0

Dblm0, 2021-01-21:
@RudyMeijer  It looks like I found a better way to prevent App autorun.
I don't know if this feature is documented somewhere, but you can tap RST button twice to do that.
The second tap should be right after the LED blue indicator shown. If successful, the blue LED indicator should stay.
Tested on Meadow.OS 0.4.4

RudyMeijer, 2021-01-21:
Hi Dblm0, Thx for this tip! In de mean time i implemented your previous tip and guess?
IT WORKS! Meadow is up and running again!

`C:\WINDOWS\system32>meadow --GetDeviceInfo
Opening port 'com4'
ID: Meadow by Wilderness Labs, Model: F7Micro, MeadowOS Version: 0.4.4 (Jan  9 2021 16:25:34), Processor: STM32F777IIK6, Processor Id: 31-00-37-00-0d-51-38-32-37-35-36-30, Serial Number: 306D35723238, CoProcessor: ESP32, CoProcessor OS Version: 0.4.4.0, Mono Version: 0.0.0.1, Device Name: MeadowF7

C:\WINDOWS\system32>meadow --MonoEnable
Opening port 'com4'
Data: Mono has been enabled. Restarting F7 Micro`

Thanks again.

bryancostanich, 2021-03-11:
@Peter-Moody i think this is similar to the not being able to disable mono problem. i think we'll need to do some investigation and thinking about this.

bryancostanich, 2021-08-08:
@RudyMeijer in b5.1 we completely rewrote the deployment toolchain. is this still an issue?

bryancostanich, 2021-09-08:
Closing after 30 days of no response.

closed 2021-09-08 by bryancostanich