WildernessLabs/Meadow_Issues #133: UdpClient fails with: Access denied

joreg, 2021-01-23:
running this code:

```
Device.InitWiFiAdapter().Wait();
if (Device.WiFiAdapter.Connect("topofthepops", "abarakadabarabistaklassahabara").ConnectionStatus != ConnectionStatus.Success)
{
     throw new Exception("Cannot connect to network, application halted.");
}
else
    Console.WriteLine("connected to wifi...");

udpClient = new UdpClient();
udpClient.Connect("192.168.0.4", 11000);
byte[] send_buffer = Encoding.ASCII.GetBytes("foo");
udpClient.Send(send_buffer, send_buffer.Length, "192.168.0.4", 5555);
```

meadow connects to the wifi successfully but then throws this error:

Unhandled Exception: System.Net.Sockets.SocketException (0x80004005): Access denied
  at System.Net.Sockets.Socket.SendTo (System.Byte[] buffer, System.Int32 offset, System.Int32 size, System.Net.Sockets.SocketFlags socketFlags, System.Net.EndPoint remoteEP) <0xc020afa0 + 0x000ea> in <27df7ff588da45c68a7fdbbfdf90ec4a>:0 
  at System.Net.Sockets.UdpClient.Send (System.Byte[] dgram, System.Int32 bytes, System.String hostname, System.Int32 port) <0xc0205100 + 0x0014a> in <27df7ff588da45c68a7fdbbfdf90ec4a>:0 
  at MeadowApplication2.MeadowApp.Initialize () <0xc150d680 + 0x000be> in <aa72e9dc6c684cc7aee717368b1f8c60>:0 
  at MeadowApplication2.MeadowApp..ctor () <0xc1023cd8 + 0x0003a> in <aa72e9dc6c684cc7aee717368b1f8c60>:0 
  at MeadowApplication2.Program.Main  System.String[] args) <0xc1023888 + 0x00034> in <aa72e9dc6c684cc7aee717368b1f8c60>:0 
 [ERROR] FATAL UNHANDLED EXCEPTION: System.Net.Sockets.SocketExce ption (0x80004005): Access denied
   at System.Net.Sockets.Socket.SendTo (System.Byte[] buffer, Sys tem.Int32 offset, System.Int32 size, System.Net.Sockets.SocketFl ags socketFlags, System.Net.EndPoint remoteEP) <0xc020afa0 + 0x0 00ea> in <27df7ff588da45c68a7fdbbfdf90ec4a>:0 
   at System.Net.Sockets.UdpClient.Send (System.Byte[] dgram, Sys tem.Int32 bytes, System.String hostname, System.Int32 port) <0xc 0205100 + 0x0014a> in <27df7ff588da45c68a7fdbbfdf90ec4a>:0 
   at MeadowApplication2.MeadowApp.Initialize () <0xc150d680 + 0x 000be> in <aa72e9dc6c684cc7aee717368b1f8c60>:0 
   at MeadowApplication2.MeadowApp..ctor () <0xc1023cd8 + 0x0003a > in <aa72e9dc6c684cc7aee717368b1f8c60>:0 
   at MeadowApplication2.Program.Main (System.String[] args) <0xc 1023888 + 0x00034> in <aa72e9dc6c684cc7aee717368b1f8c60>:0

LunicLynx, 2021-01-23:
I just had a similar issue.
For me the solution was to check that the target system is really accessible.

* Firewall turned off
* Server accepting connections on 0.0.0.0

I used another device (not meadow) to verify that it is actually working.

joreg, 2021-01-25:
@LunicLynx my understanding with UDP is that it wouldn't care about reaching the destination

seems my problem is rather this: https://github.com/WildernessLabs/Meadow_Issues/issues/123
i get a false "successful connection" and then obviously UDPClient fails..

joreg, 2021-02-16:
confirmed no longer an issue with beta4.5

closed 2021-02-16 by joreg