WildernessLabs/Meadow_Issues #134: Network performance using HttpClient is abysmal

LunicLynx, 2021-01-25:
Hello wilderness team,

first: I love meadow and things really seem to come together. So big thumbs up 👍!

Now to my issue:
I tried to send data in a loop to a server via HttpClient.
The first try does work but is already taking 10+ Seconds, the next try is taking forever to me it looks like a deadlock or something, forcing a hard reset.

Which currently blocks any further development.
I would love to help or contribute but meadow core seems to be closed source.

Is this a know issue, is there a more sophisticated example for using WiFi and HttpClient?

Thanks,
Florian

NevynUK, 2021-02-06:
Performance has been improved for the beta 4.5 release.

closed 2021-02-08 by NevynUK

LunicLynx, 2021-02-09:
Thank you. Will try it out next weekend!