WildernessLabs/Meadow_Issues #136: Bno055 not fully calibrating

SebaWnek, 2021-01-27:
In Bno055 driver:
https://github.com/WildernessLabs/Meadow.Foundation/blob/bc6b198794ed7453a346e9d012eb826c6ce9418a/Source/Meadow.Foundation.Peripherals/Sensors.Motion.Bno055/Driver/Sensors.Motion.BNO055/BNO055.cs#L1738

all 4 IsCalibrated methods use:
`return (((bno055.ReadRegister(Registers.CalibrationStatus) >> 6) & 0x03) != 0);`

The problem is fully calibrated is when corresponding registers have value 3, here value 1 is accepted as calibrated as different from 0, so without checking registers themselves there is no way to know if calibration is up to 3. maybe those could be changed to 

`return (((bno055.ReadRegister(Registers.CalibrationStatus) >> 6) & 0x03) == 3);`

so they will accept only value 3, or add public method to get calibration values, for example:

`public byte[] CheckCallibration()
{
byte calibration = bno055.ReadRegister(Registers.CalibrationStatus);
return new byte[] 
{
(byte)((callibration >> 6) & 3), 
(byte)((callibration >> 4) & 3),
(byte)((callibration >> 2) & 3),
(byte)((callibration >> 0) & 3)
}`

SebaWnek, 2021-01-27:
Proposition here: https://github.com/WildernessLabs/Meadow.Foundation/pull/144

bryancostanich, 2021-08-08:
@adrianstevens if this is still an issue, can you transfer it to Meadow.Foundation, please?