WildernessLabs/Meadow_Issues #139: Call is ambiguous issue when using Span<T> from external Library/NuGet package.

kevbite, 2021-02-09:
I'm bumped into a problem where I'm getting the following build exception when referencing another project that is netstandard referencing `System.Memory` NuGet package and uses `Span<T>` type.

> Error: /home/runner/work/Kevsoft.RTTTL/Kevsoft.RTTTL/samples/MeadowApplication/MeadowApp.cs(41,45): error CS0121: The call is ambiguous between the following methods or properties: 'System.MemoryExtensions.AsSpan(string)' and 'System.MemoryExtensions.AsSpan(string)' [/home/runner/work/Kevsoft.RTTTL/Kevsoft.RTTTL/samples/MeadowApplication/MeadowApplication.csproj]

I've pushed an example of this on to a branch here - https://github.com/kevbite/Kevsoft.RTTTL/tree/feature/meadow/samples/MeadowApplication

And you can also see the GitHub action build error here - https://github.com/kevbite/Kevsoft.RTTTL/actions/runs/552853296

jorgedevs, 2021-08-09:
hey @kevbite, following up on this issue. Is it still happening to you? I tried your sample and used your RTTTTL library and got the app working, I did had to make a few modifications on the PlayTone part in MeadowApp, so I think it should all work for you in the latest version. Nice tunes for Axel F, Batman and X Files :)

kevbite, 2021-08-09:
Hey @jorgedevs I'll check this out when I can, thanks for updating the ticket.

Can't beat some Nokia ringtone music!

kevbite, 2021-11-01:
Hey @alexischr, tried this last night with everything updated and compiled fine. Having problems with deploying to my meadow after flashing but that seems to be another story now.

Thanks for the help! I'll close off this ticket.

closed 2021-11-01 by kevbite