WildernessLabs/Meadow_Issues #140: After update to b4.5 > meadow or meadow --help wont list commands

PaulNoto, 2021-02-12:
So after I installed a perfect storm of updates yesterday 2-11-2021; Windows 10, VS2019, and Meadow b4.5;  I discovered that I can't list available commands from the Command Prompt using "meadow" or "meadow --help". Also, deploying a simple Project from VS2019 works like I have seen in the past, where it displays the Contents of the Flash, what files it is deleting, and what files it is adding. However, with my complex Project it does not show what it is changing. However, the changes are made and the new version of my app starts up with the changes. Is it just my setup, after the updates that has issues, or are these real issues with the b4.5 update? Any help would be greatly appreciated. Thanks, Paul.

PaulNoto, 2021-02-14:
Update, I still can't get meadow or meadow --help to display a list of commands.
However, My deploy issue is solved. One of the updates exposed a flaw in the way my App was structured. I redesigned the structure and all is well.

thegreatco, 2021-03-09:
@PaulNoto are you receiving any sort of error when you run `meadow --help`? Do you get any output from `meadow --version`? If so, please provide it here.