WildernessLabs/Meadow_Issues #141: HttpClient fails with 'No such host is known' when trying to POST to MSFT Oauth endpoint

beorosz, 2021-02-21:
The Meadow app crashes when trying to POST to the aforementioned server. Network package sniffer shows a DNS resolve request-response pair but the actual server is not contacted.
If I change the server name in the URL to, let's say 'wildernesslabs.co', the app contacts the WildernessLabs server and returns with HTTP 403 - so the nw communication works in this case.

Stack trace:
<pre>
Unhandled Exception: System.AggregateException: One or more errors occurred. (No such host is known) ---> System.Net.Http.HttpRequestException: No such host is known ---> System.Net.Sockets.SocketException: No such host is known
  at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) <0xc17b1b50   0x001c4> in <83417efdc24442d5bd4f09edae906397>:0 
   --- End of inner exception stack trace ---
  at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) <0xc17b1b50 + 0x002d0> in <83417efdc24442d5bd4f09edae906397>:0 
  at System.Threading.Tasks.ValueTask`1[TResult].get_Result () <0xc189d6f8 + 0x0 034> in <60f3a574daec40a7adc88cfce32b6a39>:0 
  at System.Net.Http.HttpConnectionPool.CreateConnectionAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) <0xc17aec88 + 0x00232> in <83417efdc24442d5bd4f09edae906397>:0 
  at System.Threading.Tasks.ValueTask`1[TResult].get_Result () <0xc189dac0 + 0x00034> in <60f3a574daec40a7adc88c ce32b6a39>:0 
  at System.Net.Http.HttpConnectionPool.WaitForCreatedConnectionAsync (System.Threading.Tasks.ValueTask`1[TResult] creationTask) <0xc17faac8 + 0x0013e> in <83417efdc24442d5bd4f09edae906397>:0 
  at System.Threading.Tasks.ValueTask`1[TResult].get_Result () <0xc189dac0 + 0x00034> in <60f3a574daec40a7adc88cfce32b6a39>:0 
  at System.Net.Http.HttpConnectionPool.SendWithR tryAsync (System.Net.Http.HttpRequestMessage request, System.Boolean doRequestAuth, System.Threading.CancellationToken cancellationToken) <0xc17531d8 + 0x000fa> in <83417efdc24442d5bd4f09edae906397>:0 
  at System.Net.Http.RedirectHandler.SendAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) <0xc1741578 + 0x00144> in <83417efd 24442d5bd4f09edae906397>:0 
  at System.Net.Http.HttpClient.FinishSendAsyncBuffered (System.Threading.Tasks.Task`1[TResult] sendTask, System.Net.Http.HttpRequestMessage request, System.Threading.CancellationTokenSource cts, System.Boolean disposeCts) <0xc1802038 + 0x0027c> in <83417efdc24442d5bd4f09edae906397>:0 
   --- End of inner exception stack trace ---
  at System.Threading. asks.Task.ThrowIfExceptional (System.Boolean includeTaskCanceledExceptions) <0xc18a4180 + 0x00022> in <60f3a574daec40a7adc88cfce32b6a39>:0 
  at System.Threading.Tasks.Task`1[TResult].GetResultCore (System.Boolean waitCompletionNotification) <0xc1802780 + 0x00062> in <60f3a574daec40a7adc88cfce32b6a39>:0 
  at System.Threading.Tasks.Task`1[TResult].get_Result () <0xc18026b0 + 0x000 a> in <60f3a574daec40a7adc88cfce32b6a39>:0 
  at MeadowPresenceApp.MeadowApp.AcquireAcessToken () <0xc161d638 + 0x0011a> in <254f948584a4400ca17082cad20a829f>:0 
  at MeadowPresenceApp.MeadowApp..ctor () <0xc13d0ae8 + 0x000b2> in <254f948584a4400ca17082cad20a829f>:0 
  at MeadowPresenceApp.Program.Main (System.String[] args) <0xc1023980 + 0x0003e> in <254f948584a4400ca17082cad20a8 9f>:0 
---> (Inner Exception #0) System.Net.Http.HttpRequestException: No such host is known ---> System.Net.Sockets.SocketException: No such host is known
  at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) <0xc17b1b50 + 0x001c4> in <83417efdc24442d5bd4f09edae906397>:0 
   --- End of inner e ception stack trace ---
  at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) <0xc17b1b50 + 0x002d0> in <83417efdc24442d5bd4f09edae906397>:0 
  at System.Threading.Tasks.ValueTask`1[TResult].get_Result () <0xc189d6f8 + 0x00034> in <60f3a574daec40a7adc88cfce32b6a39>:0 
  at System.Net.Http.HttpC nnectionPool.CreateConnectionAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) <0xc17aec88 + 0x00232> in <83417efdc24442d5bd4f09edae906397>:0 
  at System.Threading.Tasks.ValueTask`1[TResult].get_Result () <0xc189dac0 + 0x00034> in <60f3a574daec40a7adc88cfce32b6a39>:0 
  at System.Net.Http.HttpConnectionPool.WaitForCreatedConn ctionAsync (System.Threading.Tasks.ValueTask`1[TResult] creationTask) <0xc17faac8 + 0x0013e> in <83417efdc24442d5bd4f09edae906397>:0 
  at System.Threading.Tasks.ValueTask`1[TResult].get_Result () <0xc189dac0 + 0x00034> in <60f3a574daec40a7adc88cfce32b6a39>:0 
  at System.Net.Http.HttpConnectionPool.SendWithRetryAsync (System.Net.Http.HttpRequestMessage request, System.Boolean doR questAuth, System.Threading.CancellationToken cancellationToken) <0xc17531d8 + 0x000fa> in <83417efdc24442d5bd4f09edae906397>:0 
  at System.Net.Http.RedirectHandler.SendAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) <0xc1741578 + 0x00144> in <83417efdc24442d5bd4f09edae906397>:0 
  at System.Net.Http.HttpClient.FinishSendAs ncBuffered (System.Threading.Tasks.Task`1[TResult] sendTask, System.Net.Http.HttpRequestMessage request, System.Threading.CancellationTokenSource cts, System.Boolean disposeCts) <0xc1802038 + 0x0027c> in <83417efdc24442d5bd4f09edae906397>:0 <---
</pre>

Code for issue replication:

    public class MeadowApp : App<F7Micro, MeadowApp>
    {
        public MeadowApp()
        {
            Initialize();            
        }

        void Initialize()
        {
            Device.InitWiFiAdapter().Wait();
            var connectionResult = Device.WiFiAdapter.Connect("ssid", "password");
            if (connectionResult.ConnectionStatus != ConnectionStatus.Success)
            {
                throw new Exception("Cannot connect to network, application halted.");
            }

            var appId = "<appGuid>";
            var tenantId = "<tenantGuid>";
            var scopes = "<scopes>";

            var deviceAuthorizationRequestContent = new Dictionary<string, string>();
            deviceAuthorizationRequestContent.Add("client_id", appId);
            deviceAuthorizationRequestContent.Add("scope", scopes);
            var deviceAuthorizationRequestUri = $"https://login.microsoftonline.com/{tenantId}/oauth2/v2.0/devicecode";
            var authFormEncodedContent = new FormUrlEncodedContent(deviceAuthorizationRequestContent);

            using (HttpClient client = new HttpClient())
            {
                client.Timeout = new TimeSpan(0, 5, 0);
                var content = new StringContent(authFormEncodedContent.ReadAsStringAsync().Result, Encoding.UTF8, "application/json");
                var response = client.PostAsync(deviceAuthorizationRequestUri, content).Result;
                response.EnsureSuccessStatusCode();
                Console.WriteLine(response.Content.ReadAsStringAsync().Result);
            }
        }        
    }

bjpierre, 2021-05-17:
Adding to this to say I have had the same issues and am unable to resolve them.

The code : 
` Console.WriteLine(Dns.GetHostEntry("random.dog").AddressList[0]);
 Console.WriteLine(Dns.GetHostEntry("microsoftonline.com").AddressList[0]);`
 
 Produces the output: 
 `162.243.205.13
 Unhandled Exception: System.Net.Sockets.SocketException (0x80004005): Could not resolve host 'microsoftonline.com'
  at System.Net.Dns.Error_11001 (System.String hostName) <0xc12b01a8 + 0x0001c> in <9023935b4d11498c8a14ccc5482c13d9>:0 
  at System.Net.Dns.GetHostByName (System.String hostName) <0xc1001780 + 0x0003c> in <9023935b4d11498c8a14ccc5482c13d9>:0 
  at System.Net.Dns.GetH stEntry (System.String hostNameOrAddress) <0xc10014b8 + 0x00080> in <9023935b4d11498c8a14ccc5482c13d9>:0 
  at HelloWorld.MeadowApp.BuildToken () <0xc12af6b8 + 0x00022> in <b60e0b7df470477fa5078845194efd0a>:0 
  at HelloWorld.MeadowApp.Initialize () <0xc08b0208 + 0x0001c> in <b60e0b7df470477fa5078845194efd0a>:0 
  at HelloWorld.MeadowApp..ctor () <0xc0834698 + 0x0000a> in <b60e0b7 f470477fa5078845194efd0a>:0 
  at HelloWorld.Program.Main (System.String[] args) <0xc08344b8 + 0x00034> in <b60e0b7df470477fa5078845194efd0a>:0 `
  
  
  And that from a cell phone on the same subnet I am able to make a post to the desired endpoint, as well as run a dns lookup successfully.

NevynUK, 2021-06-08:
I have reproduced this and the problem is being investigated.

NevynUK, 2021-06-18:
`Console.WriteLine(Dns.GetHostEntry("microsoftonline.com").AddressList[0]);`

This fails in a desktop application (tested on both Mac and Windows using .NET Core 5) so I think Meadow is behaving as expected in this case.

Have you tried running the original code:

```
var appId = "<appGuid>";
var tenantId = "<tenantGuid>";
var scopes = "<scopes>";

var deviceAuthorizationRequestContent = new Dictionary<string, string>();
deviceAuthorizationRequestContent.Add("client_id", appId);
deviceAuthorizationRequestContent.Add("scope", scopes);
var deviceAuthorizationRequestUri = $"https://login.microsoftonline.com/{tenantId}/oauth2/v2.0/devicecode";
var authFormEncodedContent = new FormUrlEncodedContent(deviceAuthorizationRequestContent);

using (HttpClient client = new HttpClient())
{
    client.Timeout = new TimeSpan(0, 5, 0);
    var content = new StringContent(authFormEncodedContent.ReadAsStringAsync().Result, Encoding.UTF8, "application/json");
    var response = client.PostAsync(deviceAuthorizationRequestUri, content).Result;
    response.EnsureSuccessStatusCode();
    Console.WriteLine(response.Content.ReadAsStringAsync().Result);
}
```

as a desktop application ?

bjpierre, 2021-06-18:
Hmm, interesting, sorry about that.

I have no apps set up to use the devicecode end point currently, but to confirm that it does react differently on a desktop app vs a meadow app I put the following code on my meadow:

```

        void TestFunction()
        {
            Device.InitWiFiAdapter().Wait();
            var connectionRes = Device.WiFiAdapter.Connect(Settings.AP, Settings.APPW).Result;
            if (connectionRes.ConnectionStatus != ConnectionStatus.Success)
            {
                throw new Exception("Unable to Connect");
            }
            Console.WriteLine("WifiInited");
            var client = new HttpClient();
                var dict = new Dictionary<string, string>
                {
                    {"client_id", Settings.Cid},
                    {"client_secret", Settings.Cs},
                    {"grant_type", "client_credentials"},
                    {"scope", "offline_access https://graph.microsoft.com/.default"}
                };
                var res = client.PostAsync($"https://login.microsoft.com/{Settings.Tenant}/oauth2/v2.0/token", new FormUrlEncodedContent(dict)).Result.Content
                    .ReadAsStringAsync().Result;
                Console.WriteLine(res);
        }
```

This gave the following output:

```
Launching application...
 WifiInited
 Unhandled Exception: System.AggregateException: One or more errors occurred. (No such host is known) ---> System.Net.Http.HttpRequestException: No such host is known ---> System.Net.Sockets.SocketException: No such host is known
  at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) <0xc0f7f220   0x001c4> in <7e7719cbfda043e88c42a75a8b7871b5>:0 
   --- End of inner exception stack trace ---
  at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) <0xc0f7f220 + 0x002d0> in <7e7719cbfda043e88c42a75a8b7871b5>:0 
  at System.Threading.Tasks.ValueTask`1[TResult].get_Result () <0xc10954d8 + 0x0 034> in <fe755e761f0348d69fca7362f35b16b3>:0 
  at System.Net.Http.HttpConnectionPool.CreateConnectionAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) <0xc0f84960 + 0x00232> in <7e7719cbfda043e88c42a75a8b7871b5>:0 
  at System.Threading.Tasks.ValueTask`1[TResult].get_Result () <0xc1099948 + 0x00034> in <fe755e761f0348d69fca73 2f35b16b3>:0 
  at System.Net.Http.HttpConnectionPool.WaitForCreatedConnectionAsync (System.Threading.Tasks.ValueTask`1[TResult] creationTask) <0xc0fff170 + 0x0013e> in <7e7719cbfda043e88c42a75a8b7871b5>:0 
  at System.Threading.Tasks.ValueTask`1[TResult].get_Result () <0xc1099948 + 0x00034> in <fe755e761f0348d69fca7362f35b16b3>:0 
  at System.Net.Http.HttpConnectionPool.SendWithR tryAsync (System.Net.Http.HttpRequestMessage request, System.Boolean doRequestAuth, System.Threading.CancellationToken cancellationToken) <0xc0f6e458 + 0x000fa> in <7e7719cbfda043e88c42a75a8b7871b5>:0 
  at System.Net.Http.RedirectHandler.SendAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) <0xc0f56358 + 0x00144> in <7e7719cb da043e88c42a75a8b7871b5>:0 
  at System.Net.Http.HttpClient.FinishSendAsyncBuffered (System.Threading.Tasks.Task`1[TResult] sendTask, System.Net.Http.HttpRequestMessage request, System.Threading.CancellationTokenSource cts, System.Boolean disposeCts) <0xc10071d8 + 0x0027c> in <7e7719cbfda043e88c42a75a8b7871b5>:0 
   --- End of inner exception stack trace ---
  at System.Threading. asks.Task.ThrowIfExceptional (System.Boolean includeTaskCanceledExceptions) <0xc109a990 + 0x00022> in <fe755e761f0348d69fca7362f35b16b3>:0 
  at System.Threading.Tasks.Task`1[TResult].GetResultCore (System.Boolean waitCompletionNotification) <0xc10078b0 + 0x00062> in <fe755e761f0348d69fca7362f35b16b3>:0 
  at System.Threading.Tasks.Task`1[TResult].get_Result () <0xc1007850 + 0x000 a> in <fe755e761f0348d69fca7362f35b16b3>:0 
  at hx8357b_display.MeadowApp.TestFunction () <0xc0bcb710 + 0x00104> in <116a4ebde0de42b1b718fb34f38b44dd>:0 
  at hx8357b_display.MeadowApp..ctor () <0xc0823d58 + 0x0000a> in <116a4ebde0de42b1b718fb34f38b44dd>:0 
  at hx8357b_display.Program.Main (System.String[] args) <0xc0823b78 + 0x00034> in <116a4ebde0de42b1b718fb34f38b44dd>:0 
---  (Inner Exception #0) System.Net.Http.HttpRequestException: No such host is known ---> System.Net.Sockets.SocketException: No such host is known
  at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) <0xc0f7f220 + 0x001c4> in <7e7719cbfda043e88c42a75a8b7871b5>:0 
   --- End of inner exception st ck trace ---
  at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) <0xc0f7f220 + 0x002d0> in <7e7719cbfda043e88c42a75a8b7871b5>:0 
  at System.Threading.Tasks.ValueTask`1[TResult].get_Result () <0xc10954d8 + 0x00034> in <fe755e761f0348d69fca7362f35b16b3>:0 
  at System.Net.Http.HttpConnectionPo l.CreateConnectionAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) <0xc0f84960 + 0x00232> in <7e7719cbfda043e88c42a75a8b7871b5>:0 
  at System.Threading.Tasks.ValueTask`1[TResult].get_Result () <0xc1099948 + 0x00034> in <fe755e761f0348d69fca7362f35b16b3>:0 
[... cut ...]
                var res = client.PostAsync("http://52.191.11.87/log/{tenant}/oauth2/v2.0/token", new FormUrlEncodedContent(dict)).Result.Content
                    .ReadAsStringAsync().Result;
                Console.WriteLine(res);
        }

```

I see the following:
```
  Launching application...
   WifiInited
  {"token_type":"Bearer","expires_in":3599,"ext_expires_in":3599,"access_token":"(...)"}
 Initialize hardware...
```

If you want I can recreate exactly the original code, or I suppose beorosz can also.

Additionally I can provide my secrets values if that would make it easier for resolution of this issue!

beorosz, 2021-06-23:
@NevynUK , I run a slightly modified version of the code above as the original code gave 400 Bad Request on a desktop:

`var appId = "<appGuid>";
var tenantId = "<tenantGuid>";
var scopes = "<scopes>";

var deviceAuthorizationRequestContent = new Dictionary<string, string>();
deviceAuthorizationRequestContent.Add("client_id", appId);
deviceAuthorizationRequestContent.Add("scope", scopes);
var deviceAuthorizationRequestUri = $"https://login.microsoftonline.com/{tenantId}/oauth2/v2.0/devicecode";
var authFormEncodedContent = new FormUrlEncodedContent(deviceAuthorizationRequestContent);

using (HttpClient client = new HttpClient())
{
    client.Timeout = new TimeSpan(0, 5, 0);
    var request = new HttpRequestMessage(HttpMethod.Post, deviceAuthorizationRequestUri)
                {
                    Content = authFormEncodedContent
                };
    var response = client.SendAsync(request).Result;
    response.EnsureSuccessStatusCode();
    Console.WriteLine(response.Content.ReadAsStringAsync().Result);
}`

On the desktop it works as expected (host sends back a device code) but on Meadow it still gives the No such host is known error.

Unreal-System, 2021-08-21:
This also has a similar problem, I think the dns server should also be configurable.
I work in a university, and the network here usually cannot access the external network if the designated dns is not used.

NevynUK, 2021-08-21:
Making DNS a configurable item is on the hit list already but I have created an issue so that you can track any work on the item.

bryancostanich, 2021-09-08:
fixed in b5.3

closed 2021-09-08 by bryancostanich

Perun85, 2021-09-19:
I am experiencing the same issue with the latest version of OS and coprocessor firmware (0.5.3.0). 
It appears that none of the Azure domains can be resolved. In the test application I was trying to send device to cloud message
via HTTP to Azure IoT Hub. Unfortunately, request failed due to inability to resolve the host name.

beorosz, 2021-09-19:
I can confirm that the example above works with b5.3. After upgrading the OS to 5.3 I also had to upgrade the ESP32 and also the Mono runtime manually. What is your meadow device info output?

Perun85, 2021-09-20:
My device is able to resolve "login.microsoftonline.com" but not the DNS names of other Azure services (IoT Hub, App Service, etc.).

Everything was updated automatically using Meadow CLI. Both Meadow OS and coprocessor firmware are on version 0.5.3.0.

Perhaps I should make this comment in https://github.com/WildernessLabs/Meadow_Issues/issues/170 to emphasize the importance of configurable DNS.

NevynUK, 2021-09-20:
We are aware of the requirement to make DNS configurable and it is listed as a work item.

@Perun85 - Can you please provide a sample application that illustrates the problem you are having with Azure IoT hub.

Perun85, 2021-09-20:
To me it appears that DNS names of the services deployed in West Europe region cannot be resolved. 
Can you please try the following line of code? I will keep that hub up and running for few days.

` Dns.GetHostEntry("az220-training-hub.azure-devices.net");`

NevynUK, 2021-09-20:
Moved this to issue 173.