WildernessLabs/Meadow_Issues #142: Inconsistent Deployment errors

upswing1, 2021-03-11:
Deploying the application from VS 2019 is inconsistent.
Sometimes the deployment hangs and visual studio stops responding 
OR
The application runs differently (LED not blinking but console is displaying messages indicating that the program is running and/or program hangs at different places

upswing1, 2021-03-11:
void CycleColors(int duration)
        {
            Console.WriteLine("Cycle colors...");

            while (true)
            {
                ShowColorPulse(Color.Blue, duration);
                Echo("Blue");
                ShowColorPulse(Color.Cyan, duration);
                Echo("Cyan");
                ShowColorPulse(Color.Green, duration);
                Echo("Green");
                ShowColorPulse(Color.GreenYellow, duration);
                Echo("GreenYellow");
                ShowColorPulse(Color.Yellow, duration);
                Echo("Yellow");
                ShowColorPulse(Color.Orange, duration);
                Echo("Orange");
                ShowColorPulse(Color.OrangeRed, duration);
                Echo("OrangeRed");
                ShowColorPulse(Color.Red, duration);
                Echo("Red");
                ShowColorPulse(Color.MediumVioletRed, duration);
                Echo("MediumVioletRed");
                ShowColorPulse(Color.Purple, duration);
                Echo("Purple");
                ShowColorPulse(Color.Magenta, duration);
                Echo("Magenta");
                ShowColorPulse(Color.Pink, duration);
                Echo("Pink");
            }
        }

bryancostanich, 2021-07-12:
b5.1 saw a completely re-written Meadow.CLI and VS extension integrations which have fixed this.

closed 2021-07-12 by bryancostanich