WildernessLabs/Meadow_Issues #143: USB Device is not recognized after OS update b4.6

upswing1, 2021-03-17:
How to reproduce:
1 - Put the device into DFU Bootloader mode
2- Flash the main Meadow.OS b4.6
3- disconnect and reconnect device

Windows 10 Message => USB device not recognized and does not show on device manager. 

I flashed 2 devices with the OS b4.6 to confirm the issue and it happened on both devices.. 

Even if the device finally shows up on device manager (rarely).  The following occurs:
C:\Users\PC\AppData\Local\WildernessLabs\Meadow.CLI>meadow --MonoDisable -s COM5
Opening port 'COM5'
Error connecting to device: The semaphore timeout period has expired.

upswing1, 2021-03-17:
This occurred when updating from b4.5 to b4.6

A device that was never flashed previously behaved as expected and the update proceeded successfully.

jorgedevs, 2021-04-04:
hey @upswing1, I believe this issue has been fixed with our latest release (4.6.2). Feel free to re-open it if this issue persists.

closed 2021-04-04 by jorgedevs