WildernessLabs/Meadow_Issues #144: Calls to HttpListenerContext.Request.RemoteEndPoint result in NullReferenceException

richardjharding, 2021-03-17:
It looks like a call to Socket.RemoteEndPoint get accessor throws Null reference exception, possibly from the socket address being null?

I was able to reproduce this by modifying the Maple server sample found [here](https://github.com/WildernessLabs/Meadow.Foundation/tree/develop/Source/Meadow.Foundation.Libraries_and_Frameworks/Web.Maple/Samples/Maple.ServerSimpleMeadow_Sample)

I changed the HelloWorldRequestHandler.Hello to the following:

````
[HttpGet]
        public void Hello()
        {
            Console.WriteLine("GET::Hello");
            this.Context.Response.ContentType = ContentTypes.Application_Text;
            this.Context.Response.StatusCode = 200;
            try
            {
                var rp = this.Context.Request.RemoteEndPoint;
            }
            catch (Exception e)
            {                
                Console.WriteLine(e.Message);
                Console.WriteLine(e.StackTrace.ToString());
            }
            
            this.Send("hello world: ").Wait();
        }
````

When the request is handled and the call to RemoteEndPoint is made the following is output

```
Received GET /Hello - Invoking hello
GET::Hello
Object reference not set to an instance of an object.
  at System.Net.IPEndPoint.Create (System.Net.SocketAddress socketAddress) <0xc15496a8 + 0x00006> in <9023935b4d11498c8a14ccc5482c13d9>:0 
  at System.Net.Sockets.Socket.get_RemoteEndPoint () <0xc161c0b8 + 0x00060> in <9023935b4d11498c8a14ccc5482c13d9>:0 
  at System.Net. ttpConnection.get_RemoteEndPoint () <0xc161bff8 + 0x0000a> in <9023935b4d11498c8a14ccc5482c13d9>:0 
  at System.Net.HttpListenerRequest.get_RemoteEndPoint () <0xc161bf40 + 0x00012> in <9023935b4d11498c8a14ccc5482c13d9>:0 
  at Maple.ServerSimpleMeadow_Sample.RequestHandlers.HelloRequestHandler.Hello () <0xc161b588 + 0x00058> in <7f304e035d5c468ea4bc088025c240ab>:0 
````

This is with the latest beta 4.6 deployed

richardjharding, 2021-03-21:
Updated to 4.6.1 but the same error occurs  - in case it helps here is the log of me updating the meadow:

(note I had a couple instances of the `Opening port 'COM3'` - had to reset the board a couple times before it worked - I also repeated the whole update a second time in case something had gone wrong)

```
richa@LUGGAGE6  ~                                                                                             [12:38]
❯ meadow --Download
Download and extracted OS version 0.4.6.1 to:
C:\Users\richa\AppData\Local\WildernessLabs\Firmware
richa@LUGGAGE6  ~                                                                                             [12:39]
❯ meadow --FlashOS
Flashing OS with C:\Users\richa\AppData\Local\WildernessLabs\Firmware\Meadow.OS.bin
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Warning: Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Match vendor ID from file: 0000
Match product ID from file: 0000
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading element to address = 0x08000000, size = 2097152
Erase           [=========================] 100%      2097152 bytes
Erase    done.
Download        [=========================] 100%      2097152 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
richa@LUGGAGE6  ~                                                                                             [12:42]
❯ meadow --MonoDisable -s COM3
Opening port 'COM3'
Error connecting to device: A device attached to the system is not functioning.
richa@LUGGAGE6  ~                                                                                             [12:43]
❯ meadow --MonoDisable -s COM3
Opening port 'COM3'
Error connecting to device: A device attached to the system is not functioning.
richa@LUGGAGE6  ~                                                                                             [12:43]
❯ meadow --MonoDisable -s COM3
Opening port 'COM3'
Error connecting to device: A device attached to the system is not functioning.
richa@LUGGAGE6  ~                                                                                             [12:45]
❯
richa@LUGGAGE6  ~                                                                                             [12:45]
❯ meadow --MonoDisable -s COM3
Opening port 'COM3'
Data: Meadow will not start MONO because it didn't run correctly last time
Data: Mono enabled, will run app.exe
Data: Mono has been disabled. Restarting Meadow
Device successfully reconnected
Data: Meadow will not start MONO because it is not enabled
Data: Mono disabled, will not run app.exe
richa@LUGGAGE6  ~                                                                                             [12:45]
❯ meadow --MonoUpdateRt
Opening port 'COM3'
FileName not specified, using latest download.
Updating runtime...
[==================================================]
Data: Download of 'Meadow.OS.Runtime.bin' success (checksums calc:0xECE2E753, expected:0xECE2E753)
Data: Erasing mono flash memory
Data: Mono memory erase success
Data: Flashing 10% complete
Data: Flashing 20% complete
Data: Flashing 30% complete
Data: Flashing 40% complete
Data: Flashing 50% complete
Data: Flashing 60% complete
Data: Flashing 70% complete
Data: Flashing 80% complete
Data: Flashing 90% complete
Data: Mono runtime successfully flashed.
richa@LUGGAGE6  ~                                                                                             [12:46]
❯ meadow --FlashEsp
Opening port 'COM3'
Transferring MeadowComms.bin
Erasing ESP32 Flash...Data: Initiating ESP32 download.
done.
[==================================================]
Transferring bootloader.bin
Erasing ESP32 Flash...done.
[==================================================]
Data: File Sent Success MD5 ESP32 Calulated:'66a5f84e808ac7c6db4a1ea45cf9dab6', received from CLI:'66a5f84e808ac7c6db4a1ea45cf9dab6')
Transferring partition-table.bin
Erasing ESP32 Flash...done.
[==================================================]
Data: File Sent Success MD5 ESP32 Calulated:'d845cbc34f76d600bd979245058ddbe5', received from CLI:'d845cbc34f76d600bd979245058ddbe5')
```

bryancostanich, 2021-05-14:
This will be fixed in b5.0! Thanks @NevynUK. :)

richardjharding, 2021-05-25:
Confirmed this is fixed in b5.0

closed 2021-05-25 by richardjharding