WildernessLabs/Meadow_Issues #148: Device.SetAntenna causes a modified version of WIFI basics to hang

upswing1, 2021-03-26:
After modifying the code for Meadow.core.samples wifibasic the program hangs:

        async Task Initialize()
        {
            Console.WriteLine("Initialize hardware...");

            Device.SetAntenna(Meadow.Gateways.AntennaType.OnBoard);
            Console.WriteLine("Antenna set...");

            // initialize the wifi adpater
            if (!Device.InitWiFiAdapter().Result) {
                throw new Exception("Could not initialize the WiFi adapter.");
            }

            // connected event test.
            Device.WiFiAdapter.WiFiConnected += WiFiAdapter_ConnectionCompleted;

            // enumerate the public WiFi channels
            ScanForAccessPoints();

            // connnect to the wifi network.
            Console.WriteLine($"Connecting to WiFi Network {Secrets.WIFI_NAME}");
            var connectionResult = await Device.WiFiAdapter.Connect(Secrets.WIFI_NAME, Secrets.WIFI_PASSWORD);
            if (connectionResult.ConnectionStatus != ConnectionStatus.Success) {
                throw new Exception($"Cannot connect to network: {connectionResult.ConnectionStatus}");
            }

            Console.WriteLine($"Connected to WiFi Network {Secrets.WIFI_NAME}");

        }

Difference is the added lines: 
Device.SetAntenna(Meadow.Gateways.AntennaType.OnBoard);
            Console.WriteLine("Antenna set...");

bryancostanich, 2021-06-01:
@upswing1 is this still an issue?

NevynUK, 2021-07-05:
Closing issue as there has been no response to request for further information for more than 30 days.

closed 2021-07-05 by NevynUK