WildernessLabs/Meadow_Issues #149: RTS/CTS hardware flow control missing

upswing1, 2021-04-14:
Serial bus is currently a 2 wire Tx/Rx.  In the do do list bucket, please add support for RTS/CTS hardware flow control.

bryancostanich, 2021-05-10:
@upswing1 can you give some info on your use case? almost nothing uses RTS/CTS anymore.

upswing1, 2021-05-11:
It is a requirement for the lte feather using the 9160 nordic Semiconductor chip.  I  believe that I can work around this requirement.. I will know more soon.

ctacke, 2021-12-02:
Current Meadow does not support hardware flow control.  If a specific use-case arrives again, we can re-investigate potentially adding support on some of the GPIOs.  Closing the issue for now.

ctacke, 2024-03-26:
Re-opening. Another customer has requested this capability.

GeorgeStephens, 2024-03-26:
Connecting an RS485 transceiver directly to a meadow UART is not currently possible. For the CCM can PA15(JTDI) be optionally mapped to COM4 RTS?

ctacke, 2024-04-11:
Investigating the RTS/CTS I've come up with the following for the Core Compute module:

| Signal | Pin | Alternate Function |
| --- | --- | --- |
| UART1_CTS |  PA11 (USB_D_N) | AF7 |
| UART1_RTS | PA12 (USB_D_P) | AF7 |
| UART4_CTS | PB0 | AF8 |
| UART4_RTS | PA15/PB1 * | AF8 |  

*The ST32 docs conflict here.  The data sheet says PB1, but the AF map says PA15.  I'm inclined to believe the latter

The table above shows that the pins are brought out on the core compute module, so physically we can support RTS/CTS on either UART (COM1 or COM4).  There would be some OS work involved to set/initialize the alternate functions for those pins and to then disable them as available general purpose I/O

GeorgeStephens, 2024-04-12:
Great thanks for this! 

I'd come to the same conclusion on PA15 for UART4. We're trying to keep USB as that's kind of important for DFU mode/update... 

[image]

Peter-Moody, 2024-08-15:
The F7FeatherV2 has 2 UARTs that have RTS and CTS exposed. But, only one can be used at any time because, of the exposed pins, only D12 (PB14) is available for RTS. If you need to know about one of our other stock boards, let me know which one and I'll figure it out.
USART 3 RX - CIPO (PC11)
USART 3 TX - SCK (PC10)
USART 3 RTS - D12 (PB14)
USART 3 CTS - D06 (PB13)

UART4 RX - CIPO (PC11), D00 (PI9)
UART4 TX - SCK (PC10), D01 (PH13)
UART4 RTS - D12 (PB14)
UART4 CTS - A3 (PB0), D13 (PB15)

For the ProjLab 3.e, only UART 4 can be used.
UART 4 RX - mikroBUS 2 CIPO pin (PC11) or Grove UART connector labeled D13/COM1_TX (labeling on board is wrong), it is RX though labeled TX.
UART 4 TX - mikroBUS  2 SCK pin (PC10) or Grove UART connector labeled D12/COM1_RX (as above labeling is wrong).
UART 4 RTS - mikroBUS 2 UART_TX pin (PB14).
UART 4 CTS - mikroBUS 2 UART_AN (PB0) pin or mikroBUS 1 UART_RX pin (PB15).