WildernessLabs/Meadow_Issues #152: Meadow.CLI tries to flash ESP if mono is not disabled

bryancostanich, 2021-04-17:
check this output out:

```
bryans-MBP-3:Downloads bryancostanich$ meadow --FlashEsp
Opening port '/dev/tty.usbmodem306A356532381'
Transferring MeadowComms.bin
Erasing ESP32 Flash...Data: Mono must be disabled for ESP32 file download
done.
[==================================================]
Transferring bootloader.bin
Erasing ESP32 Flash...done.
[==================================================]
Transferring partition-table.bin
Erasing ESP32 Flash...done.
[==================================================]
```

looks like it does it, but it actually doesn't

bryancostanich, 2021-05-10:
@Peter-Moody maybe you can tackle while you're in fixing the `monodisable` bug for b5.0?

adrianstevens, 2021-07-26:
This is fixed in the new CLI

closed 2021-08-05 by NevynUK