WildernessLabs/Meadow_Issues #153: HttpClient memory leak

johncruyt, 2021-04-28:
I am using MeadowOS 4.6.2 and based on the Hackster project 'Send Data From Meadow to an ASP. NET Server over WiFi ', I want to run a HttpClient GET / POST loop every minute.
My code works perfectly, but Meadow stops working after 1 day. 
The Hackster example project, and other online HttpClient examples, only executes the GET and POST request once, so that doesn't make me any smarter.
I am using the Watchdag code to prevent Meadow from crashing.
I also checked the internal memory after each run and the free memory keeps getting smaller. Maybe a memory leak?
 
This is the code I’m running every minute:
 
            using (HttpClient client = new HttpClient())
            {
                client.Timeout = new TimeSpan(0, 5, 0);
                HttpResponseMessage response = await client.GetAsync("http://icanhazip.com/");
                try
                {
                    response.EnsureSuccessStatusCode();
                    string responseBody = await response.Content.ReadAsStringAsync();
                    responseBody = await response.Content.ReadAsStringAsync();
                }
                catch (TaskCanceledException)
                {
                    Console.WriteLine("Request time out.");
                }
                catch (Exception e)
                {
                    Console.WriteLine($"Request went sideways: {e.Message}");
                }
            }

NevynUK, 2021-06-08:
Just ran the code on 0.5.0.1 making 80,000+ requests (about 60 days worth) and the code continued to run.

Looks like this is fixed in 0.5.0.1, changing status accordingly.

johncruyt, 2021-06-09:
Great news! When the release is done I'll give it another try
Thanks!

NevynUK, 2021-07-06:
Fixed in 0.5.1.

closed 2021-07-06 by NevynUK