WildernessLabs/Meadow_Issues #154: Meadow Locks Up after 3915 Iterations

thegreatco, 2021-05-19:
```
using System;
using System.Threading;
using System.Threading.Tasks;
using Meadow;
using Meadow.Devices;
namespace PowerConsumption
{
    class Program
    {
        static PowerConsumptionApp app;
        public static async Task Main(string[] args)
        {
            if (args.Length > 0 && args[0] == "--exitOnDebug")
                return;
            // instantiate and run new meadow app
            app = new PowerConsumptionApp();
            try
            {
                await app.RunAsync()
                         .ConfigureAwait(false);
            }
            catch (Exception ex)
            {
                Console.WriteLine(ex);
            }
            Thread.Sleep(Timeout.Infinite);
        }
    }
    public class PowerConsumptionApp : App<F7Micro, PowerConsumptionApp>
    {
        public PowerConsumptionApp()
        {
        }
        public async Task RunAsync()
        {
            var counter = 0;
            MeadowOS.WatchdogEnable(10000);
            while (true)
            {
                try
                {
                    var foo = Device.GetBatteryLevel();
                    Console.WriteLine(counter);
                    counter++;
                    MeadowOS.WatchdogReset();
                }
                catch (Exception ex)
                {
                    Console.WriteLine(ex);
                    await Task.Delay(1000).ConfigureAwait(false);
                }
            }
        }
    }
}
```

Removing `Device.GetBatteryLevel()` appears to alleviate the issue.

NevynUK, 2021-06-07:
Looks like we have a fix.  Currently over 8,000 readings and still going.  Leaving the system to soak test a while longer.

NevynUK, 2021-07-06:
Fixed in 0.5.1.

closed 2021-07-06 by NevynUK