WildernessLabs/Meadow_Issues #155: Error deploying with some netstandard nuget package references (b5)

stickybun, 2021-05-25:
When referencing some netstandard nuget packages on beta 5, I start getting deployment errors.

For example add this reference to a default template in the csproj (I have projects that use this library):

```xml
<PackageReference Include="Microsoft.Extensions.Logging" Version="2.2.0" />
```

And then write some code in the MeadowApp.cs that uses a class from the library or its dependencies

```csharp
public MeadowApp()
{
    var a = new Microsoft.Extensions.Logging.Abstractions.NullLoggerFactory();
}
```

Then deploy to Meadow. On Visual Studio Mac, and VS Code Mac (haven't been able to try from Windows Visual Studio as yet),  I get the following error 

```log
Starting to Deploy to Meadow...
Error: Could not find file "/Users/user/Source/Dev/MeadowLibTest/bin/Debug/net472/System.Web.dll"
```

Or sometimes 
```log
Starting to Deploy to Meadow...
Error: Could not find file "/Users/user/Source/Dev/MeadowLibTest/bin/Debug/net472/SSystem.IO.Compression.dll"
```

DeviceInfo:
ID: Meadow by Wilderness Labs, Model: F7Micro, MeadowOS Version: 0.5.0.0 (May 23 2021 13:19:57), Processor: STM32F777IIK6, CoProcessor: ESP32, CoProcessor OS Version: 0.5.0.0, Mono Version: 0.5.0.0, Device Name: MeadowF7

richardjharding, 2021-05-26:
I think you're running into the same issue I was - that version of the logging package seems only to have been built for  NetStandard 2.0? - https://www.nuget.org/packages/Microsoft.Extensions.Logging/2.2.0

stickybun, 2021-05-26:
I can confirm, changing package to reference to 5.0.0 allows the application to deploy. And probably shows why some work, and some do not!

bryancostanich, 2021-06-01:
@adrianstevens this should be fixed with the new .NET Standard 2.1 conversion in b5.1, yeah?

stickybun, 2021-07-06:
Good news. This same repro on b5.1 appears to deploy and run ok!

closed 2021-07-07 by bryancostanich