WildernessLabs/Meadow_Issues #156: After update to beta five I'm encountering errors before my program executes.

bjpierre, 2021-05-26:
After updating to Beta 5.0 I have started encountering errors that appear to happen before my code runs. They don't cause the program to halt but they do cause visual studios onboard debugger to trigger.

The code, ignore the lack of beauty, it's a proof of concept.
```
public class MeadowApp : App<F7Micro, MeadowApp>
    {

        public MeadowApp()
        {
            WifiInit();
            Program();
        }

        void WifiInit()
        {
            Console.WriteLine("Wifi Wait");
            Device.InitWiFiAdapter().Wait();
            Console.WriteLine("Wifi Init");
            if (Device.WiFiAdapter.Connect(<>, <>).GetAwaiter().GetResult().ConnectionStatus != ConnectionStatus.Success)
            {
                throw new Exception("Unable to Connect");
            }
            Console.WriteLine("\nWifi Connected at " + Device.WiFiAdapter.IpAddress);

            using (HttpClient client = new HttpClient())
            {
                Console.WriteLine("Running Dog Test");
                HttpResponseMessage responseMessage = client.GetAsync("https://random.dog/woof.json").GetAwaiter().GetResult();
                Console.WriteLine("Response :" + responseMessage.Content.ReadAsStringAsync().GetAwaiter().GetResult());
            }
        }

        void Program()
        {
            Console.WriteLine("Hello User!\n It's time to get you signed in, are you ready?");
            //Console.ReadKey();
            Console.WriteLine("\nOkay, Lets get started");

            //Grabs a token via helper classes
            HttpClient client = new HttpClient();
            var tenant = <>;
            Dictionary<string, string> content = new Dictionary<string, string>
            {
                { "client_id", "e0e1239a-5086-46a9-84ab-fe92938d4137" },
                { "tenant", tenant },
                { "scope", "offline_access User.Read.All Presence.Read.All" }
            };
            var uri = <>;
            var res = client.PostAsync(uri, new FormUrlEncodedContent(content)).Result;
            var parsed = JsonConvert.DeserializeObject<DeviceCodeModel>(res.Content.ReadAsStringAsync().Result);
            Console.WriteLine(parsed);
            var token = new TokenAwaiter(parsed.device_code).GetToken();

            Console.WriteLine("Okay, we're signed in. Now it's time to work with this API!");

            client.DefaultRequestHeaders.Authorization = new System.Net.Http.Headers.AuthenticationHeaderValue("Bearer", token.access_token);
            res = client.GetAsync(<>).Result;
            Console.WriteLine(res.Content.ReadAsStringAsync().Result);
        }
    }
```

The error I see
```
'App.exe' (CLR v4.0.30319: App.exe): Loaded 'C:\Users\....\Meadow.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
An unhandled exception of type 'System.TypeLoadException' occurred in Unknown Module.
Non-abstract, non-.cctor method in an interface.

 Wifi Wait
 Wifi Init
 
Wifi Connected at 172.16.0.245
```

[image]

PureWeen, 2021-06-19:
@bjpierre do you know a workaround for this? I just got my chip and I'm trying to run the basic HelloMeadow app and it just crashes with this exception

bjpierre, 2021-06-20:
@PureWeen assuming you're using visual studio, don't click play, right click the solution and click deploy.

 Also, in my specific case the program was still running and console output was still being produced despite this error

jessetabali, 2022-05-02:
Still having the same issue with beta 6.3

bryancostanich, 2022-08-18:
@alexischr not sure who should look at this.

alexischr, 2022-08-19:
The original issue seems like it was simply the user clicking Play in Visual Studio instead of deploy (see https://github.com/WildernessLabs/Meadow_Issues/issues/156#issuecomment-864615421) . Is that correct @bjpierre ?

Not sure if that's the same problem you have, @jessetabali ; if so, please check out the above comment.

CartBlanche, 2022-08-25:
@jessetabali, @bjpierre could you both please specify which versions of Visual Studio you are using and what edition (Community, Professional or Enterprise)?

bjpierre, 2022-08-26:
@alexischr That is correct, it was user error as I noted in my reply to PureWeen.

@CartBlanche I believe at the time I was using Visual Studio 2019 Professional Edition.

As this was user error this could probably be closed so you guys can focus on more important things, no?

CartBlanche, 2022-08-26:
@bjpierre I'll close based on the above reply.

closed 2022-08-26 by CartBlanche