WildernessLabs/Meadow_Issues #157: BLE 5 support feature requests

wheric, 2021-05-28:
If this it not the appropriate area for this request please let me know. For the first iteration of BLE support on Meadow board I have three features that would need to be implemented in the future to allow us to integrate this into how we use BLE on android/iOS devices.

1. Client Scanning/Connection support with ability to enumerate/read/write characteristics and request the MTU size.    Without this feature one meadow f7 device cannot exchange data with another meadow device over BLE.
2. Ability to set the advertised service data 16 byte UUID and possibly a way to easily change it on the fly without tearing down the server first.  
3. Notification of an incoming characteristic read. An incoming write generates an event but a read does not from what I can tell.  Being able to respond to a read event and hand off the data to return at that moment is critical as well.   

Thank you for the effort you have put into this support so far.

bryancostanich, 2021-06-01:
@wheric this is indeed a great spot to file this, and thank you for filing the issue. we'll take this back and chat about it.

r-proto, 2021-07-08:
I agree with @wheric this is critical functionality, at the current state the meadow is only able to act as a server making it impossible to connect to other BLE devices such as sensors to read telemetry data (which is critical in my opinion in the world of IoT) or to be able to communicate with other meadow devices as @wheric mentioned. 

@bryancostanich  any chance you can provide timeline when this will be implemented?

ctacke, 2021-07-08:
Some information for planning:

1. Currently Meadow only implements a BLE server.  Client capability is on the road map, but likely not until after the 1.0 Release.  We definitely understand that it will be useful, we simply have limited resources and have other, broader features we want to make sure are delivered in v1.
2. I'll look at adding the ability to set the service UUID to a GUID and not just a ushort.  Doing it without tearing down the stack is highly unlikely.  The underlying API we're using doesn't make that easy.
3. Notifications of reads seems simple on the surface but it's not, which is why we implemented it the way it is.  The BLE radio and comms are handled by a separate processor from where your application runs, which raises the complexity of simulating interrupts over I2C.  It also would raise the latency of any client reads from the Meadow. Right now characteristic values are sent to the coprocessor on change in your app, and the coprocessor then handles reads completely independent of the main STM processor or the application.  We may add a read event/interrupt in the future, but it's unlikely to be before the v1 Release.