WildernessLabs/Meadow_Issues #159: Device.GetDeviceInformation() throws an error

brucecrombie, 2021-06-09:
Device.GetDevice.Information() throws the following error
ioctl GetDeviceInfo failed OperationNotPermitted
 Unhandled Exception: System.IndexOutOfRangeException: Index was outside the bounds of the array.
  at Meadow.Devices.F7Micro.GetDeviceInformation () <0xc0df5ce0 + 0x000f2> in <cd393da7aec048d0ac1d7ea20d381a91>:0 
  at Clima.HardwareTestbench.MeadowApp.OsTests () <0xc0df5600 + 0x00036> in <db0bad29ff6a459c98880571f3e0d965>:0 
  at Clima.HardwareTestbench.MeadowApp..ctor () <0xc0825 b8 + 0x00012> in <db0bad29ff6a459c98880571f3e0d965>:0 
  at Clima.HardwareTestbench.Program.Main (System.String[] args) <0xc0824e68 + 0x00034> in <db0bad29ff6a459c98880571f3e0d965>:0 
 [ERROR] FATAL UNHANDLED EXCEPTION: System.IndexOutOfRangeExcepti on: Index was outside the bounds of the array.
  at Meadow.Devices.F7Micro.GetDeviceInformation () <0xc0df5ce0 + 0x000f2> in <cd393da7aec048d0ac1d7ea20d381a91>:0 
  at Clima.HardwareTestbench.MeadowApp.OsTests () <0xc0df5600 + 0x00036> in <db0bad29ff6a459c98880571f3e0d965>:0 
  at Clima.HardwareTestbench.MeadowApp..ctor () <0xc08250b8 + 0x00012> in <db0bad29ff6a459c98880571f3e0d96 >:0 
  at Clima.HardwareTestbench.Program.Main (System.String[] args) <0xc0824e68 + 0x00034> in <db0bad29ff6a459c98880571f3e0d965>:0

NevynUK, 2021-07-29:
This has been fixed as part of the configuration work.

bryancostanich, 2021-09-08:
fixed in b5.3

closed 2021-09-08 by bryancostanich