WildernessLabs/Meadow_Issues #160: Meadow hangs on Device.BluetoothAdapter.StartBluetoothServer

r-proto, 2021-07-07:
When I try to start the bluetooth server it just hangs, it doesn't throw any exceptions. I have tried this on a couple meadow boards and re-flashed the devices (Updated Meadow to OS: 0.5.1.0 (Jun 19 2021 01:49:23) ESP: 0.5.1.0).

var definition = new Definition(
                "MY MEADOW",
                new Service(
                    "ServiceA",
                    253,
                    new CharacteristicString(
                        "My Text",
                        uuid: "017e99d6-8a61-11eb-8dcd-0242ac1300cc",
                        maxLength: 20,
                        permissions: CharacteristicPermission.Write | CharacteristicPermission.Read,
                        properties: CharacteristicProperty.Write | CharacteristicProperty.Read
                        )
                    )
                );
    Device.InitCoprocessor();
    try
            {
                Device.BluetoothAdapter.StartBluetoothServer(definition);
            }
            catch (Exception ex)
            {
                Console.WriteLine(ex.InnerException);
            }

greginbrisbane, 2021-07-12:
I'm running into a similar issue when using the same versions as @r-proto.

I am seeing slightly different behaviour where the device appears in BLE Scanner (iOS) but the services are not published at all.

I created a new Meadow App, used the example code from the Bluetooth documentation page and can reproduce using that basic code.  Hope that helps @ctacke.

NevynUK, 2021-07-18:
We think we have located the problem and we are working on publishing a fix.

adrianstevens, 2021-07-26:
Just an update - this has been fixed and will ship with B5.2

closed 2021-08-05 by NevynUK