WildernessLabs/Meadow_Issues #163: IByteCommunications has no WriteRegister method that takes a Span<byte> - b5.1

brucecrombie, 2021-07-18:
`WriteRegisters(byte address, byte[] data)` has been marked as obsolete but there is no new method to replace it.

There should possibly be an overload of the WriteRegister method that takes the signature
`WriteRegister(byte address, Span<byte> writeBuffer)`

This would match the existing ReadRegister method with the signature `ReadRegister(byte address, Span<byte> readBuffer)
`
This was detected during analysis of the Ds323x driver as it was throwing errors on the DisplayRegisters method. See Meadow.Foundation issue [https://github.com/WildernessLabs/Meadow.Foundation/issues/180](https://github.com/WildernessLabs/Meadow.Foundation/issues/180)

bryancostanich, 2021-08-08:
@brucecrombie thanks for the issue.

@adrianstevens we already had this as a protected method, so I made it public (`public void WriteRegister(byte address, Span<byte> writeBuffer, ByteOrder order = ByteOrder.LittleEndian)`). But I don't have the peripheral needed to test. Can you give it a whirl to validate that it's good to go, please?

See: https://github.com/WildernessLabs/Meadow.Foundation/issues/180 for more info on the test.

adrianstevens, 2021-08-09:
Absolutely!

bryancostanich, 2021-09-08:
fixed in b5.3

closed 2021-09-08 by bryancostanich