WildernessLabs/Meadow_Issues #164: VS deployment slow to copy large file

LordBenjamin, 2021-07-22:
I am downloading a large (15 MB) App.exe file to my Meadow. This is a workaround for another issue (unable to copy non-root `Content` files) in which I have added the required files to my App.exe as embedded resources.

I am observing a severe slowdown during the transfer from 60% onwards. At this point, it can take 5 - 7 minutes to complete a 10% increment. Output below, adjusted to normalise the timestamps (i.e. 00:00:00 is the start of transfer).

30 minutes to transfer 15 MB is 8.5 KB/s average. Compared to the first 10% which is around 57 KB/s (very rough numbers!)

```
ESP32 version not available. Meadow.OS version 0.5.1.0, Mono version 0.5.1.0.
...
00:00:00 Writing file: App.exe
00:00:00 Starting File Transfer...
00:00:27 Meadow StdInfo: File 10% downloaded
00:00:53 Meadow StdInfo: File 20% downloaded
00:01:19 Meadow StdInfo: File 30% downloaded
00:01:45 Meadow StdInfo: File 40% downloaded
00:02:11 Meadow StdInfo: File 50% downloaded
00:02:37 Meadow StdInfo: File 60% downloaded
00:07:16 Meadow StdInfo: File 70% downloaded
00:14:59 Meadow StdInfo: File 80% downloaded
00:22:43 Meadow StdInfo: File 90% downloaded
00:30:26 Transfer Complete, wrote 16075776 bytes to Meadow
```

LordBenjamin, 2022-04-02:
Still seeing this in latest release (tools v0.19.1).

To reproduce:

1. Create new Meadow project:
    ```
    dotnet new Meadow -n SlowCopyRepro
    ```
2. Create "large" file (e.g. 4 MB):
    ```
    fsutil file createnew SlowCopyRepro\Content01.txt 4194304
    ```
3. Update `SlowCopyRepro.csproj` to add:
    ```xml
    <ItemGroup>
        <EmbeddedResource Include="Content01.txt" />
    </ItemGroup>
    ```
4. Deploy the project to the Meadow, observing time required to transfer `App.exe`

LordBenjamin, 2022-08-09:
Thanks for the update, I'll try to give this another go at some point!

CartBlanche, 2022-08-09:
@LordBenjamin We're currently testing a fix internally which should speed up transfers a little more too. Can you confirm if you are using Mac or Windows??

LordBenjamin, 2022-08-09:
Sounds good! I'm on Windows.

LordBenjamin, 2022-08-22:
Still seems to be really slow using VS2022 / Meadow.CLI 0.19.4:

`mscorlib.dll` is ~4.42 MB and took 572 seconds to transfer (~8 KB/s).

```
2>[22/08/2022 22:48:57] Starting File Transfer...
2>[22/08/2022 22:49:05] Meadow StdInfo: File 10% downloaded
2>[22/08/2022 22:49:13] Meadow StdInfo: File 20% downloaded
2>[22/08/2022 22:50:10] Meadow StdInfo: File 30% downloaded
2>[22/08/2022 22:51:22] Meadow StdInfo: File 40% downloaded
2>[22/08/2022 22:52:33] Meadow StdInfo: File 50% downloaded
2>[22/08/2022 22:53:44] Meadow StdInfo: File 60% downloaded
2>[22/08/2022 22:54:55] Meadow StdInfo: File 70% downloaded
2>[22/08/2022 22:56:06] Meadow StdInfo: File 80% downloaded
2>[22/08/2022 22:57:17] Meadow StdInfo: File 90% downloaded
2>[22/08/2022 22:58:29] Transfer Complete, wrote 4637696 bytes to Meadow
```

Is there anything I need to do apart from `dotnet tool update -g wildernesslabs.meadow.cli` and `meadow download os; meadow flash os`?

LordBenjamin, 2022-08-22:
`meadow flash erase` did the trick (although the OS reinstall wasn't entirely smooth):

```
1>[22/08/2022 23:34:20] Writing file: mscorlib.dll
1>[22/08/2022 23:34:20] Starting File Transfer...
1>[22/08/2022 23:34:28] Meadow StdInfo: File 10% downloaded
1>[22/08/2022 23:34:35] Meadow StdInfo: File 20% downloaded
1>[22/08/2022 23:34:43] Meadow StdInfo: File 30% downloaded
1>[22/08/2022 23:34:51] Meadow StdInfo: File 40% downloaded
1>[22/08/2022 23:34:58] Meadow StdInfo: File 50% downloaded
1>[22/08/2022 23:35:06] Meadow StdInfo: File 60% downloaded
1>[22/08/2022 23:35:13] Meadow StdInfo: File 70% downloaded
1>[22/08/2022 23:35:21] Meadow StdInfo: File 80% downloaded
1>[22/08/2022 23:35:29] Meadow StdInfo: File 90% downloaded
1>[22/08/2022 23:35:37] Meadow StdInfo: Download of 'mscorlib.dll' success (checksums calculated:0x8A6F2683, expected:0x8A6F2683)
1>[22/08/2022 23:35:37] Transfer Complete, wrote 4637696 bytes to Meadow
```

~53 KB/s, consistent speed throughout the entire transfer!

CartBlanche, 2022-08-23:
Our new linker should create smaller deployable assemblies. So in an upcoming release uploads to device should, theoretically, even take less time.

CartBlanche, 2022-08-23:
@LordBenjamin are you happy for us to close this issue?

closed 2023-03-24 by alexischr