WildernessLabs/Meadow_Issues #165: Configurable Meadow WIFI NTPServer

Unreal-System, 2021-08-09:
using Device.WiFiAdapter.NtpServer I can get the time of a device.
It is a fixed item that may have limited access in different countries/regions(For example, China, the network cannot be accessed  pool.ntp.org(Because have bad [Great firewall](https://en.wikipedia.org/wiki/Great_Firewall))). It should be allowed to be modified so that the device can automatically synchronize the time after it is connected to the Internet.

Unreal-System, 2021-08-09:
Because there is a time zone error and network delay error even if there is a non-firewall problem, the NTP time will vary from place to place.

bryancostanich, 2021-08-09:
Agreed. We've been talking about this internally, and have some ideas for a better, more configurable setup here. Thanks for the issue.

closed 2021-11-08 by NevynUK