WildernessLabs/Meadow_Issues #166: SerialPort.Read method throw an exception when count greater then available chars in circular buffer

ncarandini, 2021-08-14:
The method signature is: `int Read(byte[] buffer, int index, int count)`
and the documentation says  that `count` is the maximum number of bytes to read. 

Surely enough maximum doesn't means that  `count` should be less or equal to the number of bytes in the input buffer.
Instead, maximum should mean that if `count` is greater than the number of bytes in the circular buffer then fewer bytes are read. At the end of the day, the exact number of chars read by the method is returned by the method itself.

But actually, if count is bigger that the chars available in the circular buffer, the app get stuck waiting to read count chars.

Here's the code I've found:
```
namespace Meadow.Hardware
{
  public abstract class SerialPortBase : IDisposable, ISerialPort
  {
    ...
    public int Read(byte[] buffer, int index, int count)
    {
      ...
      if (count > this._readBuffer.Count)
        throw new ArgumentException("Count cannot be larger than the available data.");
      return this._readBuffer.MoveItemsTo(buffer, index, count);
    }
  }
}
```
This should be changed to:
```
namespace Meadow.Hardware
{
  public abstract class SerialPortBase : IDisposable, ISerialPort
  {
    ...
    public int Read(byte[] buffer, int index, int count)
    {
      ...
      if (count > this._readBuffer.Count)
        count =  this._readBuffer.Count;
      return this._readBuffer.MoveItemsTo(buffer, index, count);
    }
  }
}
```

ncarandini, 2021-08-14:
BTW, the exception is not propagated to the app so the app just hang...
So this is another aspect to investigate in the future, but not related to this bug.

bryancostanich, 2022-04-24:
Upping the priority because of customer needs.

patridge, 2022-10-27:
Closing as resolved now that [RC-1 is released](http://developer.wildernesslabs.co/Meadow/Release_Notes/Release-Candidates/).

closed 2022-10-27 by patridge