WildernessLabs/Meadow_Issues #168: Don't add timestamp and other text to the Console.WriteLine() and console.Write() methods

ncarandini, 2021-08-16:
Don't add the datetime stamp and `Meadow StdOut:` message from the Console Write and WriteLine methods. It mess up all the output and make the Write method unusable.
[image]

bryancostanich, 2021-08-16:
@thegreatco @Peter-Moody does this occur in the CLI or HCOM?

thegreatco, 2021-08-16:
@bryancostanich, the timestamp is added by the `ILogger` implementation in the CLI.

Each call to `LogInformation` will be prefixed with the timestamp. If a message is sent by `HCOM` that contains multiple lines (as interpreted by the terminal), the timestamp will only show up at the beginning of the message. If I had to guess, `HCOM` is splitting the message across multiple packets. Without analyzing the source code/messages, it'll be hard to say why the messages aren't starting at the beginning of the line.

For the one line with the timestamp in the middle of a line, it is possible there is a race, though it is my understanding that `ILogger` is thread safe. I'm guessing that in this case, the issue is actually that the message spanned multiple `HCOM` messages. I thought that the `ILogger` would handle the newline character at the end of a message, but I could be wrong.

@ncarandini when you call `Console.WriteLine`, are the messages just a single line or do they possibly contain multi-line messages?

thegreatco, 2021-08-16:
The wonky formatting appears to be a side effect of the Visual Studio terminal. When run from Powershell in Windows Terminal, I'm not able to replicate the strange indentations or the timestamp appearing mid-line. When run in VS2019, I can replicate the behavior. @adrianstevens can you look into this on your end inside the extension?

adrianstevens, 2021-08-16:
The timestamp was a user contribution and it's been part of the plugin for several months - I personally have no strong preference if we want to remove it

ncarandini, 2021-08-16:
I'm using
Microsoft Visual Studio Community 2019
Version 16.11.0

Only WriteLine with single line.
You can check the source code at my GitHub project:
https://github.com/ncarandini/LoRaWAN

CartBlanche, 2022-07-04:
@ncarandini  is this still an issue with the latest versions of VS? Btw, if you still have issues, can you please provide which version of the extension you are using and what operating system and version too. Thanks.

adrianstevens, 2023-01-09:
@CartBlanche and I wonder if we should make the timestamp an option in Visual Studio? 

I personally find the time helpful but the date doesn't add any value for me. And things like "Meadow StdOut" could go as well. I believe that's done in CLI.Core.

CartBlanche, 2023-03-30:
@ncarandini With the upcoming release of RC3, we have cleaned up the Console output considerably. 
We've also split the output so it's clear which messages are coming from Visual Studio and which are being sent from Meadow. They each have their own output windows/tabs. Timestamps have been removed :). I'm going to close this issue and if you want new enhancements to the console output, please raise another issue.

closed 2023-03-30 by CartBlanche