WildernessLabs/Meadow_Issues #173: DNS Failure for Azure IoT hub services

NevynUK, 2021-09-20:
To me it appears that DNS names of the services deployed in West Europe region cannot be resolved. 
Can you please try the following line of code? I will keep that hub up and running for few days.

` Dns.GetHostEntry("az220-training-hub.azure-devices.net");`

_Originally posted by @Perun85 in https://github.com/WildernessLabs/Meadow_Issues/issues/141#issuecomment-922696838_

NevynUK, 2021-09-20:
Good news is that I have reproduced the problem.

Bad news is that I have reproduced the problem.

I am not sure this is a DNS server selection issue.  I have changed the DNS server to my local router here and that also fails even though the equivalent `nslookup` from a terminal windows on my laptop succeeds.  Also, online lookups also give information about the name including from Cloudflare (the DNS server hardcoded at the moment).

I will have a deeper look into this problem over the next few days.

I would appreciate it if you could leave the system active while I am troubleshooting.  Alternatively, let me know what you have set up so that I can try to reproduce here.  Ideally, I'd like to do that anyway so that I can add this to the test suite.

Perun85, 2021-09-21:
No problem. I can leave this IoT Hub active until this issue is resolved.  Only special thing about this one is that is deployed in West Europe region.
[image]

Few additional facts.
- this issue is also affecting app services deployed in the same region
- if you try to resolve DNS name of the IoT Hub deployed in Central US (g2cert.azure-devices.net), you will be successful

Perun85, 2021-09-22:
@NevynUK  I tested it with IoT Hub that had shorter name without hyphens (West Europe region) and it worked. 

Afterwards, I tried with "ashford-in-the-water-pc.gov.uk" and it was able to resolve that one even though it had hyphens. Perhaps it is related to the number of characters and hyphens.

NevynUK, 2021-09-22:
@Perun85 Thank you for checking that out.

I propose to leave this issue open but drop the priority as we have a work around.

Perun85, 2021-09-22:
@NevynUK Thank you for the help. It is OK with me to drop the priority.

codeputer, 2021-10-30:
Is there a status if this bug has been looked at, and if it has been found? Should I have time, can I have little information as where I should start in the public repo (which I'm assuming is public).

closed 2021-11-09 by NevynUK

NevynUK, 2021-11-09:
A buffer in the underlying RTOS needed to be expanded.  The DNS system detected there wasn't enough space for the result and sent back an error condition.