WildernessLabs/Meadow_Issues #175: Can't flash F7 v2 devices

craigktreasure, 2021-10-07:
I received my Meadow F7v2 devices a few days ago and was attempting to flash them. Flashing my first v2 device resulted in an error. After a few plug and unplugs, it seemed to boot and the LED pulsed through colors. But, I'm not sure I trust the installation after receiving an error like this as I don't think it went through the whole cycle.

I've attempted flashing from two different devices, both running Windows 11 (22000.194) with Meadow v0.14.5 and got the same result on both. I was able to successfully flash two F7v1 devices. It's just the F7v2 devices that are failing for me.

Today, on a brand new device (my new Surface Pro 8), I followed the [instructions](http://developer.wildernesslabs.co/Meadow/Getting_Started/Deploying_Meadow/):

1. Installed the .NET 5 SDK
2. Installed the driver for ST devices per the instructions [here](http://developer.wildernesslabs.co/Meadow/Getting_Started/Update_USB_Driver/).
3. Installed the Meadow CLI and downloaded the OS
    ```shell
    dotnet tool install WildernessLabs.Meadow.CLI --global
    meadow download os
    meadow flash os
    ```
4. `meadow flash os` failed with `dfu-util not found. To install, run in administrator mode: meadow install dfu-util`, so I ran `meadow install dfu-util` from an admin shell to install dfu-util 0.10.
5. Re-ran `meadow flash os` and got the following output:
```txt
Device in DFU Mode, flashing OS
Flashing OS with C:\Users\craig\AppData\Local\WildernessLabs\Firmware\Meadow.OS.bin
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 0000
Match product ID from file: 0000
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading element to address = 0x08000000, size = 2097152

Erase    done.

Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
Warning: Invalid DFU suffix signature
Device Flashed.
Connecting to Meadow on COM4
Meadow StdInfo: Meadow successfully started MONO
Meadow StdInfo: Mono is enabled
Meadow StdInfo: Mono has been disabled. Restarting Meadow
Connecting to Meadow on COM4
Meadow StdInfo: Mono is disabled.
Meadow StdInfo: Mono is disabled
Meadow StdInfo: Mono is disabled
FileName not specified, using latest download.
Starting File Transfer...
Meadow StdInfo: File 10% downloaded
Meadow StdInfo: File 20% downloaded
Meadow StdInfo: File 30% downloaded
Meadow StdInfo: File 40% downloaded
Meadow StdInfo: File 50% downloaded
Meadow StdInfo: File 60% downloaded
Meadow StdInfo: File 70% downloaded
Meadow StdInfo: File 80% downloaded
Meadow StdInfo: File 90% downloaded
Meadow StdInfo: Download of 'Meadow.OS.Runtime.bin' success (checksums calculated:0x4E2C6FF2, expected:0x4E2C6FF2)
Meadow StdInfo: Erasing mono flash memory
Exception sending command to Meadow
RequestType: HCOM_MDOW_REQUEST_MONO_UPDATE_FILE_END Timeout: 00:00:30 UserData: 1 ResponseType none CompletionMessageType: none ResponseHandler: False CompletionPredicate: True ResponsePredicate: True
Meadow.CLI.Core.Exceptions.MeadowCommandException: Command timeout waiting for response.
 ---> System.Threading.Tasks.TaskCanceledException: A task was canceled.
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.WaitForResponseMessageAsync(Command command, CancellationToken cancellationToken, String caller) in C:\WL\Meadow.CLI.Again\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:line 373
   --- End of inner exception stack trace ---
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.WaitForResponseMessageAsync(Command command, CancellationToken cancellationToken, String caller) in C:\WL\Meadow.CLI.Again\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:line 379
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.SendCommandAsync(Command command, CancellationToken cancellationToken, String caller) in C:\WL\Meadow.CLI.Again\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:line 210
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.SendTheEntireFile(FileCommand command, Boolean lastInSeries, CancellationToken cancellationToken) in C:\WL\Meadow.CLI.Again\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:line 135
Error flashing OS to Meadow
RequestType: HCOM_MDOW_REQUEST_MONO_UPDATE_FILE_END Timeout: 00:00:30 UserData: 1 ResponseType none CompletionMessageType: none ResponseHandler: False CompletionPredicate: True ResponsePredicate: True
Meadow.CLI.Core.Exceptions.MeadowCommandException: Command timeout waiting for response.
 ---> System.Threading.Tasks.TaskCanceledException: A task was canceled.
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.WaitForResponseMessageAsync(Command command, CancellationToken cancellationToken, String caller) in C:\WL\Meadow.CLI.Again\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:line 373
   --- End of inner exception stack trace ---
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.WaitForResponseMessageAsync(Command command, CancellationToken cancellationToken, String caller) in C:\WL\Meadow.CLI.Again\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:line 379
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.SendCommandAsync(Command command, CancellationToken cancellationToken, String caller) in C:\WL\Meadow.CLI.Again\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:line 210
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.SendTheEntireFile(FileCommand command, Boolean lastInSeries, CancellationToken cancellationToken) in C:\WL\Meadow.CLI.Again\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:line 135
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.UpdateMonoRuntimeAsync(String fileName, UInt32 partition, CancellationToken cancellationToken) in C:\WL\Meadow.CLI.Again\Meadow.CLI.Core\Devices\MeadowLocalDeviceFileManager.cs:line 266
   at Meadow.CLI.Core.Devices.MeadowDeviceHelper.FlashOsAsync(String runtimePath, Boolean skipRuntime, Boolean skipEsp, CancellationToken cancellationToken) in C:\WL\Meadow.CLI.Again\Meadow.CLI.Core\Devices\MeadowDeviceHelper.cs:line 371
Done!
```

craigktreasure, 2021-10-07:
From past experience, I also tried disabling mono before flashing. It didn't help.

HakanL, 2021-10-08:
I'm getting the exact same issue on my F7v2 on Windows 11. I'm able to run Hello World after this error, but not trusting the install with this error.

HakanL, 2021-10-08:
Also tested on Windows 10, with the same board. Same error.

bryancostanich, 2022-07-03:
@CartBlanche maybe we close?

closed 2022-07-04 by craigktreasure