WildernessLabs/Meadow_Issues #182: MCP23x08 throws error on constructor after update to 1.43

SebaWnek, 2021-11-08:
Hi, on: 
`expander1 = new Mcp23x08(bus, 0x20, Device.CreateDigitalInputPort(Device.Pins.D02, InterruptMode.EdgeBoth));`
In:
https://github.com/SebaWnek/WnekoTank/blob/master/WnekoTankMeadow/MeadowApp.cs line 161 (might be different as working on this repository)
It throws:
`
The method or operation is not implemented./n/n  at Meadow.Foundation.ICs.IOExpanders.Mcp23x08+I2cMcpDeviceComms.WriteRegisters (System.Byte address, System.Byte[] data) <0xc1094e38 + 0x00008> in <d341fcb26d1f426fb5c2f21afcc0b14c>:0 
  at Meadow.Foundation.ICs.IOExpanders.Mcp23x08.Initialize () <0xc1094ba0 + 0x00052> in <d341fcb26d1f426fb5c2f21afcc0b14c>:0 
  at Meadow.Foundation.[09.11.2021 00:32:14] Meadow StdOut: Cs.IOExpanders.Mcp23x08..ctor (Meadow.Foundation.ICs.IOExpanders.Mcp23x08+IMcpDeviceComms device, Meadow.Hardware.IDigitalInputPort interruptPort) <0xc108c1a0 + 0x0008e> in <d341fcb26d1f426fb5c2f21afcc0b14c>:0 
  at Meadow.Foundation.ICs.IOExpanders.Mcp23x08..ctor (Meadow.Hardware.II2cBus i2cBus, System.Byte address, Meadow.Hardware.IDigitalInputPort interruptPort) <0xc108ba50 + 0[09.11.2021 00:32:14] Meadow StdOut: 00016> in <d341fcb26d1f426fb5c2f21afcc0b14c>:0 
  at WnekoTankMeadow.MeadowApp.Initialize () <0xc0e870f0 + 0x003c6> in <4becfc35e6584c3faf225dbbdc880139>:0 
  at WnekoTankMeadow.MeadowApp..ctor () <0xc0905610 + 0x0000c> in <4becfc35e6584c3faf225dbbdc880139>:0 
[09.11.2021 00:32:16] Meadow StdOut: Unhandled Exception: System.NullReferenceException: Object reference not set to an instance of an object.
  at WnekoTankMeadow.MeadowApp..ctor () <0xc0905610 + 0x0010e> in <4becfc35e6584c3faf225dbbdc880139>:0 
  at WnekoTankMeadow.Program.Main (System.String[] args) <0xc0830ba8 + 0x00034> in <4becfc35e6584c3faf225dbbdc880139>:0 
`

After downgrading this nugget to 1.42 works again correctly

adrianstevens, 2021-11-18:
ahhh yes - sorry @SebaWnek we'll get this fixed by the next release

SebaWnek, 2021-11-19:
Thanks, good to know!

closed 2022-07-31 by adrianstevens