WildernessLabs/Meadow_Issues #183: Device becomes unresponsive after a while with HTTPS requests

pekkah, 2021-11-11:
Test app https://github.com/pekkah/meadow-libs/tree/main/src/Verify.Wifi

With previous releases and the current beta 6 the device (V2 haven't tested V1) becomes unresponsive after a while when running the app above. Previously still as related to some memory leaks which were fixed. 

NOTE: there was an custom System.dll back in the slack that had better stability (super disposal brothers)....

johncruyt, 2021-11-12:
Glad someone else is responding to this issue.
I've been raising this issue for months and there's been very little response.
This has also been addressed in #167, #179 and https://community.wildernesslabs.co/t/problem-with-repetative-http-request/1622.
Hopefully there will be a response soon and this will be resolved with priority.

pekkah, 2021-11-18:
Here's logs from latest attempt with VS extension 0.18.5

```
Meadow StdOut: Device name: VerifyWifi
Meadow StdOut: Model: F7Micro
Meadow StdOut: Processor type: STM32F777IIK6
Meadow StdOut: Product: F7Micro
Meadow StdOut: Coprocessor type: ESP32
Meadow StdOut: Coprocessor firmware version: 0.6.0.0
Meadow StdOut: Hardware revision: 2
Meadow StdOut: OS version: 0.6.0.0
Meadow StdOut: Mono version: 0.6.0.0
Meadow StdOut: Build date: Nov  4 2021 10:40:44
Meadow StdOut: Automatically connect to network: True
Meadow StdOut: Automatically reconnect: True
Meadow StdOut: Get time at startup: True
Meadow StdOut: Default access point: BASE
Meadow StdOut: Maximum retry count: 7
Meadow StdOut: Found: 31 dll files from /meadow0
Meadow StdOut: /meadow0/App.dll: 1.0.0.0
Meadow StdOut: /meadow0/netstandard.dll: 2.1.0.0
Meadow StdOut: /meadow0/mscorlib.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.Core.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.dll: 4.6.57.0
Meadow StdOut: /meadow0/Mono.Security.dll: 4.0.0.0
Meadow StdOut: /meadow0/System.Configuration.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.Xml.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.Security.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.Numerics.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.Data.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.Transactions.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.EnterpriseServices.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.Data.DataSetExtensions.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.Drawing.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.IO.Compression.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.IO.Compression.FileSystem.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.ComponentModel.Composition.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.Net.Http.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.Runtime.Serialization.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.ServiceModel.Internals.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.Web.dll: 4.0.0.0
Meadow StdOut: /meadow0/System.Runtime.Serialization.Formatters.Soap.dll: 4.6.57.0
Meadow StdOut: /meadow0/System.ComponentModel.DataAnnotations.dll: 4.6.57.0
Meadow StdOut: /meadow0/Mono.Data.Sqlite.dll: 1.0.61.0
Meadow StdOut: /meadow0/System.Xml.Linq.dll: 4.6.57.0
Meadow StdOut: /meadow0/Meadow.dll: 0.28.0.0
Meadow StdOut: /meadow0/Meadow.Contracts.dll: 0.28.0.0
Meadow StdOut: /meadow0/Meadow.Units.dll: 0.28.0.0
Meadow StdOut: /meadow0/Meadow.F7.dll: 0.28.0.0
Meadow StdOut: /meadow0/Meadow.Foundation.dll: 0.30.0.0
------------------------------------------------------------
Meadow StdOut: Connected
Before request loop. GetTotalMemory: 578KB
Request 1 11/18/2021 06:44:52
.
.
.
Meadow StdOut: Cycling...
Request 138 11/18/2021 07:02:40
Meadow StdOut: Elapsed: 2.631s, Status: OK, GetTotalMemory: 1830KB, 11/18/2021 07:02:42
Meadow StdOut: Beep...
------------------------------------------------------------
```
I added the DLL file version output to the logs if it helps.

jeffjulian, 2021-12-31:
Ditto on this issue.  Do we need a sample app for recreating the issue?

pekkah, 2021-12-31:
@jeffjulian there's an sample app at the beginning

NevynUK, 2022-04-02:
@pekkah Have you tried the 0.6.3 release ?  Just wondering if the latest stability fixes have resolved this issue.

NevynUK, 2022-08-19:
Closing issue asked for feedback over 30 days ago and no response from user..

closed 2022-08-19 by NevynUK