WildernessLabs/Meadow_Issues #186: VS deployment failures

pekkah, 2021-11-18:
Deployment reported as successful but VS build output shows this
```
Build started...
1>------ Build started: Project: Verify.Wifi, Configuration: Debug Any CPU ------
1>Verify.Wifi -> C:\projects\pekkah\meadow-libs\src\Verify.Wifi\bin\Debug\netstandard2.1\App.dll
2>------ Deploy started: Project: Verify.Wifi, Configuration: Debug Any CPU ------
2>[18/11/2021 9.07.55] Connecting to Meadow on COM7
2>[18/11/2021 9.08.02] Device has not been selected. Hit Ctrl+Shift+M to access the Device list.
2>[18/11/2021 9.08.02]      Deploy failed: Object reference not set to an instance of an object.
2>[18/11/2021 9.08.02]      Reset Meadow and try again                                                                                               (00:00:00.0001466 since last.)
========== Build: 1 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
========== Deploy: 1 succeeded, 0 failed, 0 skipped ==========
```
Clicking deploy again work properly. So most of the time have to click deploy twice. Also the success report on the bottom toolbar is annoying as I need to double check the build output to make sure it actually worked.

Meadow extension version: 0.18.5. Tried uninstall/install but error persists.

adrianstevens, 2021-11-18:
Thanks @pekkah - interesting that it works the 2nd time 

Have you tried resetting Meadow before deploying? I'm curious if its serial port issue or a Meadow issue.

pekkah, 2021-11-19:
Hmm.
Got some new logs
```
Build started...
2>------ Deploy started: Project: Verify.Wifi, Configuration: Debug Any CPU ------
2>[] Connecting to Meadow on COM7
2>[] Meadow StdOut: Beep...
2>  
2>[] Device has not been selected. Hit Ctrl+Shift+M to access the Device list.
2>[]     Deploy failed: Object reference not set to an instance of an object.
2>]     Reset Meadow and try again                                                                                               (00:00:00.0002980 since last.)
========== Build: 1 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
========== Deploy: 1 succeeded, 0 failed, 0 skipped ==========
```
That "Beep.." there after connecting to Meadow message is actually the heartbeat thread I have running on the device. So it must connect to it but then fail with that null reference exception.

Resetting before deployment helps.

bryancostanich, 2022-07-03:
@alexischr this is some HCOM stuff, yeah? I wonder if the thread priority stuff @ctacke was talking about in Slack will help responsiveness here.

adrianstevens, 2023-01-09:
@pekkah are you seeing an improvement with the latest version of the Meadow tools?

@CartBlanche @alexischr - I believe we can close the issue

Peter-Moody, 2023-03-24:
Deploy is working reliably at this time.

closed 2023-03-24 by Peter-Moody