WildernessLabs/Meadow_Issues #190: ADCs seems slower than the data sheet implies they should be

ctacke, 2021-12-18:
The STM data sheet implies the ADC can sample at 5 Ms/s:
https://www.stm32duino.com/viewtopic.php?t=138

Simple code to read fast gets about 4 samples/sec

```
            var ln = Device.CreateAnalogInputPort(Device.Pins.A00, 1);
            var lf = Device.CreateAnalogInputPort(Device.Pins.A01, 1);
            var rn = Device.CreateAnalogInputPort(Device.Pins.A02, 1);
            var rf = Device.CreateAnalogInputPort(Device.Pins.A03, 1);
...
private void TestSpeed()
{
    var start = Environment.TickCount;

    // read the analogs
    var ln = await _leftNear.Read();
    var lf = await _leftFar.Read();
    var rn = await _rightNear.Read();
    var rf = await _rightFar.Read();

    var et = Environment.TickCount - start;
    Console.WriteLine($"Started at {start} for an ET of {et}");

}
```

alexischr, 2022-10-06:
@ctacke I think the Read() method on time-sensitive interaces should perhaps not be async at all

Peter-Moody, 2023-10-19:
This is resolved with Issue #342 (19Oct23)

closed 2024-03-19 by adrianstevens