WildernessLabs/Meadow_Issues #196: Debugging Meadow App in Windows 10

command0r, 2022-01-05:
I'm familiarizing myself with some sample projects provided on the website (HelloPulsy particularly). The board is assembled and the app works as expected when deployed from VS. However, any attempts to run the same app in a debug mode fail. The app does reach a breakpoint (see below):

[image]
In the console I see the following message:
``` 
Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'netstandard, Version=2.1.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51' or one of its dependencies. The system cannot find the file specified.

D:\Meadow\HelloPulsy\MeadowApplication\bin\Debug\netstandard2.1\App.dll (process 22308) exited with code -532462766.
To automatically close the console when debugging stops, enable Tools->Options->Debugging->Automatically close the console when debugging stops.
Press any key to close this window . . .
```
The output in the 'output' window is a bit different:
```
The target process exited without raising a CoreCLR started event. Ensure that the target process is configured to use .NET Core. This may be expected if the target process did not run on .NET Core.
The program '[22308] App.dll: Program Trace' has exited with code 0 (0x0).
The program '[22308] App.dll' has exited with code 3762504530 (0xe0434352).
Meadow StdOut: Initializing file system...
Meadow StdOut: File system initialized.
Meadow StdOut: Initialize hardware...
```
The environment is the following:
- Windows 10 (21H1)
- VS 2022 with Meadow extension and 'Mobile development' workload **installed**

Also, it's not clear how debugging works. Is there an emulator for Meadow F7? Where can I find some documentation on debugging with Meadow board and VS? 
All I was able to find here http://developer.wildernesslabs.co/Meadow/Getting_Started/Hello_World is a quick mention to include the **Mobile Development with .NET** component when installing Visual Studio, so you can debug.

command0r, 2022-01-06:
Another curious observation. I found out that you can initiate a remote debugging session on the device (Meadow F7) by using the 'meadow debug' CLI command. I assume all you need to do is to connect VS debugger to a remote process (127.0.0.1:4027). This doesn't work as well. Moreover, what I see in the console is the following:
```
Meadow StdInfo: Mono is disabled
Meadow StdInfo: Mono has been enabled. Restarting F7 Micro
Connecting to Meadow on COM4
Meadow StdInfo: Mono will not start. The following file is missing:Meadow.dll
Meadow StdInfo: Mono is enabled
Connecting to Meadow on COM4
Meadow StdInfo: Mono will not start. The following file is missing:Meadow.dll
Listening for Visual Studio to connect on 127.0.0.1:4024
```
**Meadow.dll** is missing. Where is it supposed to come from? Isn't it the part of the OS?
I tried to wipe out the OS and the file system and redo everything from scratch - the result is the same.

Guys, I'm an experienced engineer with years of experience, but I don't have to guess how it all works. I can only imagine what it's like for the newbies. The absence of proper documentation is killing, as well as a dead silence in this repository.

t9mike, 2022-02-05:
I have also just started on Hello World and running into issues.

With Visual Studio 2019, I can run the cycle color sample. And if I change to only two colors I see the change deployed. But eventually VS just crashes and restarts. VS reports that the issue is the extension. So the extension is crashing. Breakpoints are never hit.

So I tried running in VS 2022. There I see the same error you do and it does not even deploy.

This is with both versions of VS 100% up-to-date and brand new install of the Meadow extension.

bryancostanich, 2022-02-05:
@adrianstevens @CartBlanche

CartBlanche, 2022-02-05:
Hi @command0r and @t9mike, sorry about the late reply. Could you please specify exactly which versions of both VS and Meadow extension you are using. Also confirm which version of the Meadow board you have (V1 or V2) and I'll try to repro the issue on my Windows machine.

t9mike, 2022-02-05:
I just realized an issue with my setup. I had updated VS2019 and 2022 with all MS updates. I installed Meadow extension on 2019. There I ran into the issue with deploy working but debugging not working and the extension eventually crashing. Not @command0r error.

So I switched over to 2022. **But I did not install the 2022 extension**. This is where I saw @command0r error. But it really did appear the extension was installed and my my sleepy brain did not notice lack of extension install. Are there shared bits between VS versions?

I just installed 2022 Meadow extension and ran. It eventually crashes. Issue may be ignoring the com port setting in some places. My board is at COM5:

[image]

But when I run I see mix of COM5 and COM3 in log:

```
1>------ Deploy started: Project: MeadowHelloWorld, Configuration: Debug Any CPU ------
1>[2/5/2022 8:30:37 AM] Connecting to Meadow on COM5
1>[2/5/2022 8:30:39 AM] Download version file for release 0.6.0.0
1>[2/5/2022 8:30:39 AM] Downloaded 285 bytes     
1>[2/5/2022 8:30:39 AM] OS version 0.6.0.0 is already installed.
1>[2/5/2022 8:30:39 AM] Meadow StdInfo: Mono is enabled
1>[2/5/2022 8:30:39 AM] Meadow StdInfo: Mono has been disabled. Restarting Meadow
1>[2/5/2022 8:30:43 AM] Connecting to Meadow on COM5
1>[2/5/2022 8:30:45 AM] Meadow StdInfo: Mono is disabled
1>[2/5/2022 8:30:45 AM] Meadow StdInfo: dev/     
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  little0 [block]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  little0p0 [block]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  monostderr [char]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  monostdout [char]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  mtdblock0 [block]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  null [char]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  nxupd [char]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  pwm0 [char]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  ramlog [char]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  ttyACM0 [char]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  ttyS0 [char]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  ttyS1 [char]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  ttyS2 [char]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  upd [char]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  urandom [char]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  usrsock [char]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo: meadow0/ 
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  ./      
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  ../     
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  App.exe [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  App.pdb [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  App.deps.json [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  netstandard.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  mscorlib.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Core.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  Mono.Security.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Configuration.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Xml.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Security.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Numerics.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Data.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Transactions.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.EnterpriseServices.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Data.DataSetExtensions.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Drawing.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.IO.Compression.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.IO.Compression.FileSystem.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.ComponentModel.Composition.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Net.Http.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Runtime.Serialization.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.ServiceModel.Internals.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Web.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Runtime.Serialization.Formatters.Soap.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.ComponentModel.DataAnnotations.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  Mono.Data.Sqlite.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  System.Xml.Linq.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  Meadow.dll [file]
1>[2/5/2022 8:30:45 AM] Meadow StdInfo:  Meadow.Contracts.dll [file]
[... 197 lines cut ...]

VisualStudio.Mac   1.0
Mac Extension for Visual Studio

Xamarin   17.0.0.343 (d17-0@fb07a17)
Visual Studio extension to enable development for Xamarin.iOS and Xamarin.Android.

Xamarin Designer   17.0.0.183 (remotes/origin/d17-0@a351f0f1f)
Visual Studio extension to enable Xamarin Designer tools in Visual Studio.

Xamarin Templates   17.0.17 (9e779b0)
Templates for building iOS, Android, and Windows apps with Xamarin and Xamarin.Forms.

Xamarin.Android SDK   12.1.0.5 (d17-0/6b0e6b2)
Xamarin.Android Reference Assemblies and MSBuild support.
    Mono: c633fe9
    Java.Interop: xamarin/java.interop/d17-0@febb1367
    ProGuard: Guardsquare/proguard/v7.0.1@912d149
    SQLite: xamarin/sqlite/3.36.0@a575761
    Xamarin.Android Tools: xamarin/xamarin-android-tools/d17-0@a5194e9

Xamarin.iOS and Xamarin.Mac SDK   15.4.0.0 (8fc41ae82)
Xamarin.iOS and Xamarin.Mac Reference Assemblies and MSBuild support.
```
[image]

command0r, 2022-02-06:
Hi @CartBlanche, here's the data you're looking for:
VS 2022, v17.04
Extensions:
- Meadow.F7 v0.29.1
- Meadow.Foundation v0.31.0

t9mike, 2022-02-06:
I have V2 board.

command0r, 2022-02-09:
@adrianstevens,  @CartBlanche, @bryancostanich, do you guys have any insights into this issue? I see it wasn't even labelled. I don't mind promptly helping you with any information you may request from my end, but I'd expect a reply a bit more often than once in a month.

bryancostanich, 2022-02-09:
@command0r sorry about the tardiness of the reply. we've been head's down on trying to get b6.2 out the door which has a pile of fixes around this workflow, and so we don't always get to the issues as quickly as we should.

that said, we have a [slack server](http://slackinvite.wildernesslabs.co/) and if you pop into the #support channel, you'll find that you can usually get realtime support pretty quickly. so if you run into things needing urgent attention, I 100% recommend that over the meadow_issues repo.

t9mike, 2022-02-10:
In https://github.com/WildernessLabs/Meadow_Issues/issues/196#issuecomment-1030637068 I have screen shot showing COM5 is selected (this is my port). But in places the deploy prints COM3 to screen: 

1>[2/5/2022 8:30:53 AM] Connecting to Meadow on COM5
1>[2/5/2022 8:30:54 AM] Connecting to Meadow on COM3 <-- WHY???

[image]

No hard coded COM3 lurking in code? That was my guess.

I am in USA Central time zone.

CartBlanche, 2022-02-10:
@t9mike COM ports aren't/shouldn't be hard coded. Do you have more than one device plugged in, hence why 2 COM ports are found? The only other thing I can think of it that someone the device is moving from one COM port to another.

I'm on UK/GMT time. Are you around in 1hour 15mins? That would make it by 16:30 so about 10:30 your time (or 11:30 DC time)?

CartBlanche, 2022-02-10:
@t9mike Thanks for taking the time to investigate further and identifying what could potentially be a USB-C issue (??!!, very odd). I don't believe we are doing anything fancy with USB connector types. If you do find out anything, we'd love to know about it. I don't have a USB-C port on either of my laptops (too old) to test this, but wonder if @bryancostanich or @adrianstevens do?

CartBlanche, 2022-02-10:
@command0r I'm just updating my Windows VS2022 setup to 17.0.6 and I'll retest the latest Meadow extension and get back to you.

CartBlanche, 2022-02-11:
@command0r I've managed to repro that dialog. As we've not changed anything in our extension and only VS2022 update has happened, I suspect that the VS team may have broken something in debugging, but I need to investigate further.

CartBlanche, 2022-02-12:
@command0r I was able to work around this issue by uninstalling the Meadow extension, then a manually doing **flash os**  on my v2 board. I think reinstalled the 0.18.7 Meadow extension. After that that everything worked in both VS2019 and VS2022 for me and has continued to work.

command0r, 2022-02-13:
@CartBlanche, I just applied the workaround you suggested and can confirm it worked for me. It makes me puzzled what was wrong the first time, but the app finally hits the breakpoint and deploys to the F7 board.

CartBlanche, 2022-02-14:
@command0r The only thing that occurs to me is that the original flash os didn't complete successfully and left the device in an unstable state, which then cause the extension to hand during debugging. We'll have to see if we are getting any messages back from the device to VS itself to make this a more graceful experience.

CartBlanche, 2022-02-14:
Resolved. Workout provided.

closed 2022-02-14 by CartBlanche