WildernessLabs/Meadow_Issues #198: Neopixel / ws2812 driver.

NevynUK, 2022-01-06:
Recording investigation of the Neopixel timing for future reference.

The Neopixels (WS2812) LEDs use a time dependent signal to differentiate between a 0 and a 1.  There is no clock for synchronisation as there is with I2C or SPI.  The timings look something like this:

[image]

The reset signal is a single low pulse greater than 50 microseconds (T<sub>reset</sub>).

For the remaining signals the timings are:

Signal | Time (nanoseconds)
------|----
TOH | 400
T0L | 800
T1H | 850
T1L | 450

All times have a tolerance of 150 nanoseconds giving the minimum and maximum timings of:

Signal | Min | Max
------|----|----
TOH | 250 | 550
T0L | 650 | 950
T1H | 700 | 1000
T1L | 300 | 600

Use of the SPI interface to use the COPI data line as a driver for the LEDs has been investigated and some issues were noted.  By changing the clock speed it should be possible to change the CPOI output to mimic the timing signal the Neopixels are expecting.

Given the tolerances a simple approach would be to adopt the following timings:

Signal | Time (nanoseconds)
------|----
TOH | 400
T0L | 800
T1H | 800
T1L | 400

So if we assume a single bit takes 200 nanoseconds to transmit then the number of bits becomes:

Signal | Number of bits
------|----
TOH | 2 (x 1)
T0L | 4 (x 0)
T1H | 4 (x 1)
T1L |  2 (x 0)

The next thing to consider is the time between the bytes.

The approach above was tried but did not work.  

Firstly, the data within the bytes was transmitted was OK, the problem was the time delay between the bytes.  This was consistently greater than the maximum time for T1H.

The second problem that was noted concerned the transmission of a repeated pattern.  A repeated pattern of 20 bytes of 0xA1 resulted in the first four bytes being transmitted as expected, i.e. the end of the 1 in 0xA1 dropped to low at the end of the clock for the byte.  It then raised to 1 at the start of the next byte.  The problem comes with bytes 5 and 6 (and the following bytes).  The data signal remains high at the end of byte 5 through to the start of byte 6.  Strictly speaking this is allowed but it messes up the theory of operation for this problem.

adrianstevens, 2022-01-06:
This is awesome - for visibility - I attempted to brute-force control of a single NeoPixel on a GPIO pin with no luck

```
void SendNeoBit(bool isOne)
        {
            long ticks = DateTime.Now.Ticks;

            if(isOne)
            {
                neoPixel.State = true;
                while (DateTime.Now.Ticks - ticks < 8000) { }
                neoPixel.State = false;
                while (DateTime.Now.Ticks - ticks < 4500 + 8000) { }
            }
            else
            {
                neoPixel.State = true;
                while (DateTime.Now.Ticks - ticks < 4000) { }
                neoPixel.State = false;
                while (DateTime.Now.Ticks - ticks < 8500 + 4000) { }
            }
        }
```

UeberDaniel, 2022-07-23:
I have the same problem. I have tried something around and with a simple toggle function to get the maximum frequency and I get an exact frequency of 1 kHz with the GPIO pins switching. This is much too little for this application...

UeberDaniel, 2022-07-23:
The Solution is to use SPI:

I got it working using this values for Low and High bits and this function to convert the Color struct:

```
static byte lowBit = 0x7E;
static byte highBit = 0x70;

public static void ColorToLedData(Color input, out byte[] ledData)
        {
            ledData = new byte[24];
            var bits = new BitArray(new byte[] { input.R, input.B, input.G } );

            for (int i = 0; i < 24; i++)
            {
                if (bits.Get(i)) ledData[i] = highBit ;
                else ledData[i] = lowBit ;
            }
        }
```

Maybe the MSB/LSB orientation is wrong atm but its a first hint.

bryancostanich, 2023-06-07:
Update: Dominick Caponi got this to work. Now up to them and @adrianstevens to bash a driver together.

dcaponi, 2023-06-08:
for reference - https://www.hackster.io/blaker/ws2812-leds-in-c-on-the-meadow-df8acf

UeberDaniel, 2023-06-10:
Please build a driver that can support multiple led types (e.g. for SK8612 RGBW LEDs). I am currently working on a driver for WS2812b display panels (display with variable count of panels and size). For data transfer I used this one so far: [https://github.com/UeberDaniel/PIC16F18325_WS2812bLEDDriver](url).
I will include your new driver in the display driver as soon as it is ready.

UeberDaniel, 2023-06-13:
WS2812b Display driver is now working using MicroGraphics:

display = new WS2812bDisplay(Device.CreateSpiBus(), 16, 16, 1, 1);

[image]

display = new WS2812bDisplay(Device.CreateSpiBus(), 8, 8, 2, 2);

[image]

display = new WS2812bDisplay(Device.CreateSpiBus(), 8, 8, 4, 1);

[image]

display = new WS2812bDisplay(Device.CreateSpiBus(), 8, 8, 1, 4);

[image]

UeberDaniel, 2023-07-04:
Driver should be working: [, https://github.com/WildernessLabs/Meadow.Foundation/issues/504](https://github.com/UeberDaniel/Ws2812Display)

Tested with 4 x 8x8, 1 x 16x16 and 4 x 32x8 panels.
Would be great if you can add it to the Meadow.Fundation repository.

johnkattenhorn, 2023-07-12:
I think if my memory serves I saw this in the release notes for 1.0.4, right @adrianstevens ?