WildernessLabs/Meadow_Issues #202: Extension crashes deploying app w/ large number of large assemblies

t9mike, 2022-02-06:
Not sure if related to https://github.com/WildernessLabs/Meadow_Issues/issues/196#issuecomment-1030637068 (I don't think so because only COM5 is referenced, the correct port), but I decided to throw a bunch of our library projects at a sample app to see what execution time is for a beefy function call. But the extension crashes during deploy at the same point every time. I can't try a deploy again without resetting the board.
```
1>------ Deploy started: Project: MeadowApplication1, Configuration: Debug Any CPU ------
1>[2/5/2022 9:11:47 PM] Connecting to Meadow on COM5
1>[2/5/2022 9:11:47 PM] Meadow StdInfo: Meadow successfully started MONO
1>[2/5/2022 9:11:47 PM] Meadow StdErr: 
1>         
1>[2/5/2022 9:11:47 PM] Meadow StdErr: Unhandled Exception:
1>
1>[2/5/2022 9:11:47 PM] Meadow StdErr: System.IO.FileNotFoundException: Could not load file or assembly
1>[2/5/2022 9:11:47 PM] Meadow StdErr:  'Com.SiLabs.Timing.Device.Si55xx, Version=4.2.0.0, Culture=neut
1>[2/5/2022 9:11:47 PM] Meadow StdErr: ral, PublicKeyToken=null' or one of its dependencies.
1>
1>[2/5/2022 9:11:47 PM] Meadow StdErr: File name: 'Com.SiLabs.Timing.Device.Si55xx, Version=4.2.0.0, Cu
1>[2/5/2022 9:11:47 PM] Meadow StdErr: lture=neutral, PublicKeyToken=null'
1>  at MeadowApplication1.Program.Main (System.String[] args) <0xc0832fb8 + 0x00034> in <1c44045c748b4e0f861863d0236c4672>:0 
1>[ERROR] FATAL UNHANDLED EXCEPTION: System.IO.FileNotFoundException: Could not load file or assembly 'Com.SiLabs.Timing.Device.Si55xx, Version=4.2.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies.
1>File n
1>[2/5/2022 9:11:47 PM] Meadow StdErr: me: 'Com.SiLabs.Timing.Device.Si55xx, Version=4.2.0.0, Culture=neutral, PublicKeyToken=null'
1>  at MeadowApplication1.Program.Main (System.String[] args) <0xc0832fb8 + 0x00034> in <1c44045c748b4e0f861863d0236c4672>:0 
1>
1>[2/5/2022 9:11:48 PM] OS download failed, make sure you have an active internet connection
1>[2/5/2022 9:11:48 PM] Meadow StdInfo: Mono is enabled
1>[2/5/2022 9:11:48 PM] Meadow StdInfo: Mono has been disabled. Restarting Meadow
1>[2/5/2022 9:11:52 PM] Connecting to Meadow on COM5
1>[2/5/2022 9:11:54 PM] Meadow StdInfo: Mono is disabled
1>[2/5/2022 9:11:54 PM] Meadow StdInfo: dev/     
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  little0 [block]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  little0p0 [block]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  monostderr [char]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  monostdout [char]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  mtdblock0 [block]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  null [char]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  nxupd [char]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  pwm0 [char]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  ramlog [char]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  ttyACM0 [char]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  ttyS0 [char]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  ttyS1 [char]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  ttyS2 [char]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  upd [char]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  urandom [char]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo:  usrsock [char]
1>[2/5/2022 9:11:54 PM] Meadow StdInfo: meadow0/ 
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  ./      
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  ../     
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  App.exe [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  App.pdb [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  App.deps.json [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  netstandard.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  mscorlib.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.Core.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  Mono.Security.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.Configuration.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.Xml.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.Security.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.Numerics.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.Data.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.Transactions.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.EnterpriseServices.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.Data.DataSetExtensions.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.Drawing.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.IO.Compression.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.IO.Compression.FileSystem.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.ComponentModel.Composition.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.Net.Http.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.Runtime.Serialization.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.ServiceModel.Internals.dll [file]
1>[2/5/2022 9:11:55 PM] Meadow StdInfo:  System.Web.dll [file]
[... 147 lines cut ...]
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): System.ComponentModel.Composition.dll
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): System.Net.Http.dll
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): System.Runtime.Serialization.dll
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): System.ServiceModel.Internals.dll
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): System.Web.dll
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): System.Runtime.Serialization.Formatters.Soap.dll
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): System.ComponentModel.DataAnnotations.dll
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): Mono.Data.Sqlite.dll
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): System.Xml.Linq.dll
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): Meadow.dll
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): Meadow.Contracts.dll
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): Meadow.Units.dll
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): Meadow.F7.dll
1>[2/5/2022 9:12:06 PM] Skipping file (hash match): Meadow.Foundation.dll
1>[2/5/2022 9:12:06 PM] Writing file: Com.SiLabs.Timing.Device.Si55xx.dll
1>[2/5/2022 9:12:06 PM] Starting File Transfer...
1>[2/5/2022 9:12:59 PM] Meadow StdInfo: File 10% downloaded
1>[2/5/2022 9:13:52 PM] Meadow StdInfo: File 20% downloaded
1>[2/5/2022 9:14:45 PM] Meadow StdInfo: File 30% downloaded
1>[2/5/2022 9:15:38 PM] Meadow StdInfo: File 40% downloaded
1>[2/5/2022 9:16:32 PM] Meadow StdInfo: File 50% downloaded
1>[2/5/2022 9:17:27 PM] Meadow StdInfo: File 60% downloaded
```

CRASH!

CartBlanche, 2022-07-04:
@t9mike  is this still an issue with the latest versions of VS? Btw, if you still have issues, can you please provide which version of the extension you are using and what operating system and version too. Also if you have a sample project that demonstrates the issue, that would help. Thanks.

t9mike, 2022-07-04:
Hello @CartBlanche . I am not actively using Meadow, so feel free to close.

closed 2022-07-04 by t9mike