WildernessLabs/Meadow_Issues #203: Visual studio 2022 restarts while debugging

upswing1, 2022-02-23:
I get the following error while debugging a meadow's app
Meadow StdInfo: Mono is enabled
Connecting to Meadow on COM4
Meadow StdInfo: Meadow successfully started MONO
An error occurred while listening for debugging connections

and visual studio 2022 closes and restarts..

Next time that I run the app it goes further but stops at 
Error receiving data from Visual Studio

upswing1, 2022-07-05:
> @upswing1<https://github.com/upswing1> is this still an issue with the latest versions of VS? Btw, if you still have issues, can you please provide which version of the extension you are using and what operating system and version too. Thanks.

I am on vacation and I cannot test it at this time. As soon as I come back I will update the status.  However I remember the error as happening consistently. I will be able to test in about 3 weeks

CartBlanche, 2022-08-25:
@upswing1 Have you had a chance to re-try with latest versions of VS?

CartBlanche, 2022-08-26:
@upswing1 we are close to releasing v1. I'll close your issues for now, feel free to reopen them if you see them post V1.

closed 2022-08-26 by CartBlanche