WildernessLabs/Meadow_Issues #205: Cannot step through code in debug mode

upswing1, 2022-02-24:
Once a breakpoint is reached. I cannot step through the code because VS hangs with following message Error receiving data from Visual Studio

CartBlanche, 2022-07-04:
@upswing1 is this still an issue with the latest versions of VS? Btw, if you still have issues, can you please provide which version of the extension you are using and one what operating system. Thanks.

upswing1, 2022-07-05:
I am on vacation and I cannot test it at this time. As soon as I come back I will update the status.  However I remember the error as being a hard error.

CartBlanche, 2022-08-25:
@upswing1  Bump to find out if. you've had a chance to try this will the latest Meadow extensions?
Could you also confirm which Edition of VS you are using (Community, Professional or Enterprise) ?

upswing1, 2022-08-25:
I have not tested the latest Meadows extensions yet.  When I reported the error I was using VS community 2019.

I am waiting for the first release version (Not Beta) to resume testing.  I believe that it should come out soon.

I am sorry that I am not of more help today

Jack

CartBlanche, 2022-08-26:
@upswing1 yes we are close to releasing v1. I'll close your issues for now, feel free to reopen them if you see them post V1.

closed 2023-03-30 by CartBlanche