WildernessLabs/Meadow_Issues #207: Device.Connect never retruns

PaulGostelow, 2022-04-01:
When I make a call to connect to wifi:

`var connectionResult = await Device.WiFiAdapter.Connect(ssid, password);`

The Connect never returns if you give it the wrong password (I say never, I left it for 20 minutes); the correct password connected in 30s. (edited)

patridge, 2022-08-24:
For some additional notes, this appears to happen in a couple scenarios.

* Trying to connect to a non-existent network
* Trying to connect to an existing-but-5GHz network

Neither of these should be supported, I guess, the non-existent network for obvious reasons and the 5GHz network due to ESP hardware limitations. It would be a much nicer situation for developers to work around these situations, whether accidental or not, if it threw an exception of some kind, even if it's just a timeout.

patridge, 2022-08-24:
And for some code reference, when connecting to either a 5GHz or non-existent network, this code will not log anything currently, neither the connection status error nor some other caught exception.

```csharp
try
{
    var connectionResult = await wifi.Connect(Secrets.WIFI_NAME, Secrets.WIFI_PASSWORD, TimeSpan.FromSeconds(45));

    if (connectionResult.ConnectionStatus != ConnectionStatus.Success)
    {
        throw new Exception($"Cannot connect to network: {connectionResult.ConnectionStatus}");
    }
    else
    {
        DisplayNetworkInformation();
    }
}
catch (Exception ex)
{
    Console.WriteLine($"Failed to connect to Wi-Fi: {ex}");
    throw;
}
```

NevynUK, 2023-01-04:
Fixed in RC2

closed 2023-01-04 by NevynUK

patridge, 2023-01-04:
I can confirm here, thank you! Looks like it now throws a `NetworkException` when this fails in code, in case anyone is curious.

I'm not seeing any diagnostic information when the wifi.config.yaml network doesn't exist, but I'll look into it more and file a new issue if necessary.