WildernessLabs/Meadow_Issues #208: Old version of app.exe executed when uploaded file exceeds a certain size

LordBenjamin, 2022-04-02:
Follow-on from #164.

I noticed that when I uploaded a "large" `App.exe` (due to embedded resources) that the Meadow board did not execute the new file - instead, the previous version was executed (obvious due to differences in console messages).

I have not yet determined the threshold, but based on quick experimentation expect it to fall between 2 MB (still working) and 4 MB (not working).

All output messages indicate file transfer success, the deploy completes and the Meadow reboots without incident. However, the running version of the application is not correct.

bryancostanich, 2022-08-09:
@CartBlanche this should be working now with linking, yeah?

CartBlanche, 2022-08-09:
@bryancostanich in theory is should be a lot better, but it will also depend just how large @LordBenjamin 's exe is my with the embedded resources. I've not test his specific scenario with the latest build yet.

alexischr, 2023-02-07:
@LordBenjamin we haven't heard back if linking has solved your "running out of space" issue? I assumed that the file write was failing because of space.

LordBenjamin, 2023-02-07:
@alexischr sorry, I've not had time to do anything with the Meadow for a while. I wouldn't say it was space related as I could reproduce with a file under 4 MB.

CartBlanche, 2023-03-30:
@LordBenjamin I've just tested this with a 2MB image, which is both copied to the device and is also and embedded resource, using our internal RC3 build and everything appears to work as it should. I'll close this for now, but feel free to re-open if you hit this issue again. Maybe send us a zip file of the project that breaks.

closed 2023-03-30 by CartBlanche