WildernessLabs/Meadow_Issues #211: Device crash after a while with HTTP requests

tinohager, 2022-06-24:
I received my test board today and immediately started a small test project. After several http requests the app crashes (67-68 in my case)

> Start request...
[24.06.2022 13:31:07] Meadow StdOut: request done
Start request...
[24.06.2022 13:31:10] Meadow StdOut: System.Net.Http.HttpRequestException: An error occurred while sending the request. ---> System.IO.IOException: Unable to read data from the transport connection: Not enough buffer space is available. ---> System.Net.Sockets.SocketException: Not enough buffer space is available
   --- End of inner exception stack trace ---
  at System.Net.Sockets.Socket+AwaitableSocketAsyncEventArg[24.06.2022 13:31:10] Meadow StdOut: .ThrowException (System.Net.Sockets.SocketError error) <0xc13dc1d8 + 0x0000c> in <8a9e2547771049b7b67cf85ccc325795>:0 
  at System.Net.Sockets.Socket+AwaitableSocketAsyncEventArgs.GetResult (System.Int16 token) <0xc1270c00 + 0x0003e> in <8a9e2547771049b7b67cf85ccc325795>:0 
  at System.Threading.Tasks.ValueTask`1[TResult].get_Result () <0xc1270af0 + 0x0004e> in <cda647e009ac4d28b0[24.06.2022 13:31:10] Meadow StdOut: 2a10df00f6f96>:0 
  at System.Net.Http.HttpConnection.FillAsync () <0xc12652b0 + 0x001e8> in <9f3755425dde4cea824b72a811fdf90e>:0 
  at System.Net.Http.HttpConnection.ReadNextResponseHeaderLineAsync (System.Boolean foldedHeadersAllowed) <0xc1263ff8 + 0x00418> in <9f3755425dde4cea824b72a811fdf90e>:0 
  at System.Threading.Tasks.ValueTask`1[TResult].get_Result () <0xc1274810 + 0x000[24.06.2022 13:31:10] Meadow StdOut: 4> in <cda647e009ac4d28b042a10df00f6f96>:0 
  at System.Net.Http.HttpConnection.SendAsyncCore (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) <0xc1245e88 + 0x0178e> in <9f3755425dde4cea824b72a811fdf90e>:0 
   --- End of inner exception stack trace ---
  at System.Net.Http.HttpConnection.SendAsyncCore (System.Net.Http.HttpRequestMe[24.06.2022 13:31:10] Meadow StdOut: sage request, System.Threading.CancellationToken cancellationToken) <0xc1245e88 + 0x020ae> in <9f3755425dde4cea824b72a811fdf90e>:0 
  at System.Net.Http.HttpConnectionPool.SendWithNtConnectionAuthAsync (System.Net.Http.HttpConnection connection, System.Net.Http.HttpRequestMessage request, System.Boolean doRequestAuth, System.Threading.CancellationToken cancellationToken) <0xc12045[24.06.2022 13:31:10] Meadow StdOut: 0 + 0x00224> in <9f3755425dde4cea824b72a811fdf90e>:0 
  at System.Net.Http.HttpConnectionPool.SendWithRetryAsync (System.Net.Http.HttpRequestMessage request, System.Boolean doRequestAuth, System.Threading.CancellationToken cancellationToken) <0xc1148790 + 0x00264> in <9f3755425dde4cea824b72a811fdf90e>:0 
  at System.Net.Http.RedirectHandler.SendAsync (System.Net.Http.HttpRequestMe[24.06.2022 13:31:10] Meadow StdOut: sage request, System.Threading.CancellationToken cancellationToken) <0xc1128d30 + 0x00144> in <9f3755425dde4cea824b72a811fdf90e>:0 
  at System.Net.Http.HttpClient.FinishSendAsyncUnbuffered (System.Threading.Tasks.Task`1[TResult] sendTask, System.Net.Http.HttpRequestMessage request, System.Threading.CancellationTokenSource cts, System.Boolean disposeCts) <0xc11df9e0 + 0x00126> in [24.06.2022 13:31:10] Meadow StdOut: 9f3755425dde4cea824b72a811fdf90e>:0 
  at RgbLedSample.MeadowApp.ConnectHttpTest () <0xc1094560 + 0x00116> in <2a995760a2274694b88960bea36ae3ea>:0 
[24.06.2022 13:31:10] Meadow StdOut: Start request...
[24.06.2022 13:31:17] Meadow StdOut: poll_event_wait: mono_poll () failed, error (118) Error code '118'

**test code**
```cs
RgbLed rgbLed;

public MeadowApp()
{
	Initialize().Wait();
	ConnectHttpTest().Wait();
}

async Task Initialize() 
{
	rgbLed = new RgbLed(
		device: Device,
		redPin: Device.Pins.OnboardLedRed,
		greenPin: Device.Pins.OnboardLedGreen,
		bluePin: Device.Pins.OnboardLedBlue
	);

	Console.WriteLine("Connect WLAN...");
	var connectionResult = await Device.WiFiAdapter.Connect("SSID", "Password");
	if (connectionResult.ConnectionStatus != ConnectionStatus.Success)
	{
		for (var i = 0; i < 10; i++)
		{
			rgbLed.SetColor(RgbLed.Colors.Red);
			Thread.Sleep(200);
			rgbLed.SetColor(RgbLed.Colors.Black);
			Thread.Sleep(200);
		}
		throw new Exception($"Cannot connect to network: {connectionResult.ConnectionStatus}");
	}
	else
	{
		for (var i = 0; i < 10; i++)
		{
			rgbLed.SetColor(RgbLed.Colors.Green);
			Thread.Sleep(50);
			rgbLed.SetColor(RgbLed.Colors.Black);
			Thread.Sleep(50);
		}
	}
}

async Task ConnectHttpTest()
{
	using var httpClient = new HttpClient();

	while (true)
	{
		try
		{
			Console.WriteLine("Start request...");
			using var response = await httpClient.GetAsync("http://webhook.site/xxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx", HttpCompletionOption.ResponseHeadersRead);
			Console.WriteLine("request done");
			if (response.IsSuccessStatusCode)
			{
				rgbLed.SetColor(RgbLed.Colors.Green);
			}
			else
			{
				rgbLed.SetColor(RgbLed.Colors.Red);
			}
		}
		catch (Exception exception)
		{
			Console.WriteLine(exception);
			Thread.Sleep(200);
		}
	}
}
```

NevynUK, 2022-06-29:
Currently trying to reproduce this.

tinohager, 2022-06-30:
Great thank you!

tinohager, 2022-07-20:
@NevynUK some news available?

NevynUK, 2022-07-26:
We have found out what is happening on the ESP32, now working on the why so that we can resolve the issue.

NevynUK, 2022-08-16:
@tinohager Application current runs for over 200 requests so I'm marking the issue as fixed in the next release.

bmrudolph, 2022-09-07:
@NevynUK is there an update on this release.  This appears to happen on HttpListener, sockets, etc as well. It has me at a stand still without any reliable networking.

patridge, 2022-10-27:
Closing as resolved now that [RC-1 is released](http://developer.wildernesslabs.co/Meadow/Release_Notes/Release-Candidates/).

closed 2022-10-27 by patridge