WildernessLabs/Meadow_Issues #213: WiFi Adapter Cannot Connect using 0.6.4.0. HttpListener stops responding.

JohnBoyWilson, 2022-07-17:
This app will not connect. The program stalls at the line:

Device.WiFiAdapter.Connect(Secrets.WIFI_NAME, Secrets.WIFI_PASSWORD).Wait(); (MeadowApp.cs Line 326)

Using 0.6.2.0 it will connect. However, HandleIncomingConnects (MeadowApp.cs Line 468) stops responding after it serves a few pages. I had to simplify the web pages that it's serving because certain HTML text would cause it to crash.

Secrets has been scrubbed.

[Listener.zip](https://github.com/WildernessLabs/Meadow_Issues/files/9128750/Listener.zip)

NevynUK, 2022-07-20:
Reproduced, under investigation.

NevynUK, 2022-07-25:
@JohnBoyWilson Can you please try swapping the use of pins D02 and D03 to see if that gets you going.

So the button is in input on D03 and the RGB LED is an output on D02.

This may get you moving again.

JohnBoyWilson, 2022-07-25:
I will try that and let you know how it goes. Can you explain why you think this will make a difference?

NevynUK, 2022-07-25:
The minimum reproduction we have is:

```cs
protected IDigitalInterruptPort button = null;
protected IDigitalOutputPort redLED = null;

public MeadowApp()
{
    //
    //  D02 as an output port works OK, the connection request succeeds.
    //
    redLED = Device.CreateDigitalOutputPort(Device.Pins.D02);
    //
    //  D02 as an input port causes the connection to fail
    //
    //button = Device.CreateDigitalInputPort(Device.Pins.D02);

    Console.WriteLine("Connecting to access point.");
    Device.WiFiAdapter.Connect(Secrets.WIFI_NAME, Secrets.WIFI_PASSWORD).Wait();
    Console.WriteLine("WiFi connection completed.");
}
```

We are still investigating the root cause and happy to let you know the results when we have a firm conclusion.

JohnBoyWilson, 2022-07-27:
Switching the pins is working for me. Makes me think I should avoid all the multi-function pins.

NevynUK, 2022-07-27:
It is OK to use the multifunction pins as documented.  The only things to be aware of are the PWM timer group and the interrupt groups. 

This particular issue was caused by a conflict in the interrupt group for D02, it uses group 10 which is used by the STM/ESP comms.  Group 10 should have been reserved for the STM/ESP comms but due to an oversight it was not.  This will be fixed in the next release.  It will mean that in future D02 can be used as an input without interrupts.

JohnBoyWilson, 2022-07-30:
FYI - pins 14 and 15 cause the same problems. Put output to red led back on D3. Then try to put the button input on D14 or D15. Both of those hang. Have to go down to D13 for it to work.

NevynUK, 2022-07-30:
Yes, fixed as part of the review from the original report.  SPI ClK should also be avoided as an interrupt port on the V2 boards.

The pin out diagrams are also being updated to reflect this.

closed 2023-08-29 by adrianstevens