WildernessLabs/Meadow_Issues #216: HTTP Listener stops responding after a few requests

JohnBoyWilson, 2022-08-17:
Starts up and runs fine. After a few requests, the HTTP Listener stops responding. Had to simplify the HTML to get it to respond at all
[HenHouse.zip](https://github.com/WildernessLabs/Meadow_Issues/files/9366327/HenHouse.zip)
.

bryancostanich, 2022-08-18:
@NevynUK might be fixed with your new work.

LordBenjamin, 2022-08-22:
I have also been trying to figure out intermittent hangs where `HttpListener.GetContextAsync()` / `HttpListener.GetContext()` did not return even though I had made a HTTP request. Great if this is fixed!

bryancostanich, 2023-05-26:
@NevynUK this is fixed now, yeah?

NevynUK, 2023-09-27:
Updated the application to current SDK:

[MeadowApp.cs.zip](https://github.com/WildernessLabs/Meadow_Issues/files/12737113/MeadowApp.cs.zip)

Deployed the application and ran several `curl` commands and managed to get to 1100+ requests with commands such as:

```zsh
repeat 400 { date; curl -s http://192.168.1.15:8081/ -o /dev/null }
```

Eventually hit:

```
Meadow StdOut: UnhandledException
Meadow StdOut: System.IO.IOException: Unable to write data to the transport connection: Connection reset by peer.
Meadow StdOut: at System.Net.Sockets.NetworkStream.EndWrite (System.IAsyncResult asyncResult) [0x0006e] in <e675d9f2be98419ab899a5aae534c835>:0
Meadow StdOut: at System.Net.ResponseStream.EndWrite (System.IAsyncResult ares) [0x00051] in <e675d9f2be98419ab899a5aae534c835>:0
Meadow StdOut: at System.IO.Stream+<>c.<BeginEndWriteAsync>b__58_1 (System.IO.Stream stream, System.IAsyncResult asyncResult) [0x00000] in <77563d
Meadow StdOut: 372d948198f425b41a9b84c90>:0
Meadow StdOut: at (wrapper delegate-invoke) System.Func`3[System.IO.Stream,System.IAsyncResult,System.Threading.Tasks.VoidTaskResult].invoke_TResult_T1_T2(System.IO.Stream,System.IAsyncResult)
Meadow StdOut: at System.Threading.Tasks.TaskFactory`1+FromAsyncTrimPromise`1[TResult,TInstance].Complete (TInstance thisRef, System.Func`3[T1,T2,TResult] endMethod, System.IAsyncResult
Meadow StdOut: syncResult, System.Boolean requiresSynchronization) [0x00000] in <77563d7372d948198f425b41a9b84c90>:0
Meadow StdOut: --- End of stack trace from previous location where exception was thrown ---
Meadow StdOut: at HttpListener_Basics.MeadowApp+<>c__DisplayClass58_0.<HandleIncomingConnections>b__0 () [0x00396] in <47e3f5f67a794a109d983afbc860e4fd>:0
Meadow StdOut: at HttpListener_Basics.MeadowApp.HandleIncomingConnection
Meadow StdOut: () [0x0009c] in <47e3f5f67a794a109d983afbc860e4fd>:0
Meadow StdOut: at HttpListener_Basics.MeadowApp.StartServer () [0x000b2] in <47e3f5f67a794a109d983afbc860e4fd>:0
Meadow StdOut: at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__7_1 (System.Object state) [0x00000] in <77563d7372d948198f425b41a9b84c90>:0
Meadow StdOut: at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context (S
Meadow StdOut: stem.Object state) [0x00007] in <77563d7372d948198f425b41a9b84c90>:0
Meadow StdOut: at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <77563d7372d948198f425b41a9b84c90>:0
Meadow StdOut: at System.Threading.ExecutionContext.Run (System.Threading.E
Meadow StdOut: ecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <77563d7372d948198f425b41a9b84c90>:0
Meadow StdOut: at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00021] in <77563d7372d948198f425b41a9b84c90>:0
Meadow StdOut: at System.Threading.ThreadPoolWorkQueue.Dispat
Meadow StdOut: h () [0x00074] in <77563d7372d948198f425b41a9b84c90>:0
Meadow StdOut: at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <77563d7372d948198f425b41a9b84c90>:0
Meadow StdOut: Inner System.Net.Sockets.SocketException: Connection reset by peer
Meadow StdOut: at System.Net.Sockets.Socket.EndSend (System.IAsyncResult asyncResult) [0x00012] in <e675d9f2be98419ab899a5aae534c835>:0
Meadow StdOut: at System.Net
Meadow StdOut: Sockets.NetworkStream.EndWrite (System.IAsyncResult asyncResult) [0x0002a] in <e675d9f2be98419ab899a5aae534c835>:0
Meadow StdOut: Shutdown
```

bryancostanich, 2023-10-17:
Looks like a bug in `SendTo()` on the ESP32 side. Mark in progress on debugging.

NevynUK, 2023-11-30:
Memory leak (31 bytes) was found and corrected.  Initial tests have shown that the number of requests has increased from ~850 to over 4,200.

Longer soak testing in progress.

NevynUK, 2023-11-30:
Longer soak test has passed 11,000 requests and is still running.

Marking as fixed.