WildernessLabs/Meadow_Issues #219: Wi-Fi connection hangs on repeat deployments

patridge, 2022-08-31:
[Originally discovered by @ctacke]

There are times when we can reproduce this problem without issue. And then there are times when I can't get it to happen at all.

## Steps

1. Create a Meadow app that connects to a wi-fi network.

    ```csharp
    public override async Task Run()
    {
        var wifi = Device.NetworkAdapters.Primary<IWiFiNetworkAdapter>();
        wifi.NetworkConnected += WiFiAdapter_NetworkConnected;
        Console.WriteLine($"Connecting to WiFi Network {Secrets.WIFI_NAME}");
        var connectionResult = await wifi.Connect(Secrets.WIFI_NAME, Secrets.WIFI_PASSWORD, TimeSpan.FromSeconds(45));
    }

    void WiFiAdapter_NetworkConnected(object sender, EventArgs e)
    {
        Console.WriteLine($"Connection request completed. Address: {(sender as INetworkAdapter)?.IpAddress}");
    }
    ```

1. Right-click the Meadow app project in Solution Explorer and select **Deploy**.
1. Wait for the app to log successful network connection event handler calls (twice, which is a separate issue).
1. Before doing anything else, repeat the right-click > **Deploy** step.

## Follow-up efforts

I reproduced this for most of a couple hours. Then, while trying to find more edge cases, I could no longer reproduce the issue at all, even with the same code above. I don't know if the following was a result of no longer reproducing the issue or not, but it appeared that the `connectionResult` every time. I can't remember now if that was pointing to the connection succeeding without the `NetworkConnected` event firing.

## Potential workarounds

There are times when using a different approach or doing various things between deployments will avoid this connection hang.

* If you change any code after step 1, it seems to connect fine.
* If you do Debug > Start Debugging (or F5) instead of the Deploy option, it seems to connect fine repeatedly.
* If you reset the Meadow manually between step 1 and 2, it seems to connect fine the second time.
* It's possible that the connection event isn't firing even though the connection is successful as described above.

## Environment

* Windows 10 Pro 22H2: 19045.1949
* Visual Studio 2022 Enterprise: 17.3.1
* VS 2022 Tools for Meadow [Preview]: v0.19.6

patridge, 2022-08-31:
As a weird additional possibly-related, possibly-new-issue edge case, connecting to a second network within the same app run will also not fire the `NetworkConnected` event the second time unless you explicitly call `Disconnect` on the adapter first.

ctacke, 2022-09-09:
When logging WiFi events from the coprocessor, when it succeeds, I see this:
```
[9/8/2022 8:14:13 PM] Meadow StdOut: INFORMATION: WiFi Event: StartWiFiInterfaceEvent
[9/8/2022 8:14:13 PM] Meadow StdOut: INFORMATION: WiFi Event: ConnectToAccessPointEvent
```
When it hangs, I see this:
```
[9/8/2022 8:12:16 PM] Meadow StdOut: INFORMATION: WiFi Event: StartWiFiInterfaceEvent
[9/8/2022 8:12:16 PM] Meadow StdOut: INFORMATION: WiFi Event: DisconnectFromAccessPointEvent
```

NelsonLamprecht, 2022-10-24:
I'm not sure if the issue I've been seeing is related but:
(IP address is statically assigned)

- Meadow is powered off for some time (over a day)
- Meadow is powered on
- Meadow device is pingable within a second or two
- Meadow device goes through other initiation
- Meadow device stops being pingable
- Meadow device is rebooted via reset button
- Meadow device is pingable within a second or two
- Meadow device goes through other initiation
- Meadow device behaves normally