WildernessLabs/Meadow_Issues #220: Wi-Fi NetworkConnected fired twice per new connection

patridge, 2022-08-31:
When connecting to a Wi-Fi network, the connection event handler seems to get called twice per connection, once for IP address `0.0.0.0` and once for the final IP address provided by the network.

## Steps

1. Create simple Wi-Fi connection app similar to a reduced version of the [Meadow.Core.Samples WiFi_Basics app](https://github.com/WildernessLabs/Meadow.Core.Samples/blob/develop/Source/Meadow.Core.Samples/Network/WiFi_Basics/MeadowApp.cs) with your network SSID and password in a Secrets.cs class.

    ```csharp
    public override async Task Run()
    {
        var wifi = Device.NetworkAdapters.Primary<IWiFiNetworkAdapter>();
        wifi.NetworkConnected += WiFiAdapter_NetworkConnected;
        Console.WriteLine($"Connecting to Wi-Fi Network {Secrets.WIFI_NAME}");
        var connectionResult = await wifi.Connect(Secrets.WIFI_NAME, Secrets.WIFI_PASSWORD, TimeSpan.FromSeconds(45));
    }
    
    void WiFiAdapter_NetworkConnected(object sender, EventArgs e)
    {
        Console.WriteLine($"Connection request completed. Address: {(sender as INetworkAdapter)?.IpAddress}");
    }
    ```
1. Run this app and observe the output.

```console
Meadow StdOut: Connecting to WiFi Network [redacted]
Meadow StdOut: Connection request completed. Address: 0.0.0.0
Meadow StdOut: Connection request completed. Address: 192.168.0.168
```

## Environment

* Windows 10 Pro 22H2: 19045.1949
* Visual Studio 2022 Enterprise: 17.3.1
* VS 2022 Tools for Meadow [Preview]: v0.19.6
* Meadow OS: v.0.6.7.11
* Meadow projects latest from `develop` as of 30 August 2022

patridge, 2022-08-31:
Edited to reflect that `NetworkDisconnected` also exhibits this behavior in case the fix is related.

ctacke, 2022-09-09:
Verified that multiple events are coming into Core from the ESP.

patridge, 2022-10-27:
It appears this is half-fixed. It is fixed in `NetworkConnected` but _not_ fixed in `NetworkDisconnected` yet.

While verifying two consecutive network connections in #221 ([comment](https://github.com/WildernessLabs/Meadow_Issues/issues/221#issuecomment-1292903537)), I didn't see duplicate connection logging, but I do see double disconnect logging. Here's the 

```console
Launching application...

[2022-10-26 20:53:33] Meadow StdOut: INFORMATION: Update Service is disabled.
[2022-10-26 20:53:33] Meadow StdOut: INFORMATION: Meadow OS v.
[2022-10-26 20:53:33] Meadow StdOut: Connecting to WiFi Network {network1}
[2022-10-26 20:53:38] Meadow StdOut: NetworkConnected - Address: 192.168.0.168
[2022-10-26 20:53:39] Meadow StdOut: 
wlan0
=====
  Adapter name: wlan0
  Interface type .......................... : Wireless80211
  Physical Address ........................ : ...
  Operational status ...................... : Up
  IP version .............................. : IPv4
  MTU ..................................... : 0
  IP address .............................. : 192.168.0.168
[2022-10-26 20:53:39] Meadow StdOut:   Subnet mask ............................. : 255.255.255.0
[2022-10-26 20:53:39] Meadow StdOut: connectionResult: Success
[2022-10-26 20:53:39] Meadow StdOut: wifi.IpAddress after await Connect: 192.168.0.168
[2022-10-26 20:53:44] Meadow StdOut: wifi.IpAddress after delay: 192.168.0.168
[2022-10-26 20:53:50] Meadow StdOut: Requesting https://postman-echo.com/get?foo1=bar1&foo2=bar2 - 10/27/2022 02:53:44
[2022-10-26 20:54:09] Meadow StdOut: Success [length: 247]
[2022-10-26 20:54:10] Meadow StdOut: wifi.IpAddress after disconnect: 192.168.0.168
[2022-10-26 20:54:10] Meadow StdOut: NetworkDisconnected
[2022-10-26 20:54:10] Meadow StdOut: NetworkDisconnected
[2022-10-26 20:54:15] Meadow StdOut: wifi.IpAddress after disconnect+delay: 192.168.0.168
Connecting to WiFi Network {network2}
[2022-10-26 20:54:16] Meadow StdOut: NetworkConnected - Address: 192.168.0.168
[2022-10-26 20:54:16] Meadow StdOut: 
wlan0
=====
  Adapter name: wlan0
  Interface type .......................... : Wireless80211
  Physical Address ........................ : ...
  Operational status ...................... : Up
  IP version .............................. : IPv4
  MTU ..................................... : 0
  IP address .............................. : 192.168.0.168
[2022-10-26 20:54:17] Meadow StdOut:   Subnet mask ............................. : 255.255.255.0
connectionResult: Success
wifi.IpAddress after await Connect: 192.168.0.168
[2022-10-26 20:54:22] Meadow StdOut: wifi.IpAddress after delay: 192.168.0.168
Requesting https://postman-echo.com/get?foo1=bar1&foo2=bar2 - 10/27/2022 02:54:22
[2022-10-26 20:54:24] Meadow StdOut: Success [length: 247]
```

NevynUK, 2022-11-23:
Fixed in RC1-1

closed 2022-11-23 by NevynUK