WildernessLabs/Meadow_Issues #222: [Meadow.CLI] OS download failures can prevent successful flash

patridge, 2022-09-02:
If there is an error while downloading some part of a new OS version, an error will be logged, but subsequent calls to download will not fix the missing elements.

## Expected

If an OS download element fails, it might be worth discarding what was downloaded, or holding it as a partial download for future attempts.

Without that, on a second `download` call, the CLI might verify that all the expected components are downloaded so that any follow-up `meadow flash os` call has all the components it needs. Without the missing components, some components will possibly install fine until the first missing component fails.

## Actual

The missing files will be gone from the OS download and a second attempt to download will say the version is already "installed (sic)" (different bug with PR: WildernessLabs/Meadow.CLI#183). There doesn't appear to be a way to delete the partially-downloaded OS version from within Meadow.CLI either.

### Example output

This output is caused by a missing ESP32 firmware on the server, but the outcome is the same. That file is not present in the downloaded OS, but the download system doesn't know that for a second call

```console
> meadow download os --osVersion 0.6.7.13
Download version file for release 0.6.7.13
Downloaded 288 bytes
Downloading latest MCU firmware
Downloaded 4.95 MiB
Downloaded 9.77 MiB
Downloaded 12.74 MiB
Downloaded 15.41 MiB
Downloading latest ESP32 firmware
ERROR
System.Net.Http.HttpRequestException: Response status code does not indicate success: 404 (Not Found).
  at System.Net.Http.HttpResponseMessage.EnsureSuccessStatusCode()
  at Meadow.CLI.Core.DownloadManager.DownloadFileAsync(Uri uri, CancellationToken cancellationToken) in D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\DownloadManager.cs:265
  at Meadow.CLI.Core.DownloadManager.DownloadAndExtractFileAsync(Uri uri, String target_path, CancellationToken cancellationToken) in D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\DownloadManager.cs:280
  at Meadow.CLI.Core.DownloadManager.DownloadLatestAsync(String version, Boolean force) in D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\DownloadManager.cs:139
  at Meadow.CLI.Commands.Utility.DownloadOsCommand.ExecuteAsync(IConsole console) in D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI\Commands\Utility\DownloadOsCommand.cs:27
  at CliFx.CliApplication.RunAsync(ApplicationSchema applicationSchema, CommandInput commandInput) in D:\a\CliFx\CliFx\CliFx\CliApplication.cs:147
  at CliFx.CliApplication.RunAsync(IReadOnlyList`1 commandLineArguments, IReadOnlyDictionary`2 environmentVariables) in D:\a\CliFx\CliFx\CliFx\CliApplication.cs:191
Done!
> meadow download os --osVersion 0.6.7.13
Download version file for release 0.6.7.13
Downloaded 288 bytes
OS version 0.6.7.13 is already installed.
Done!
```

## Workaround

By adding `--force` to the `download` command, users can try again from scratch. Alternatively, if a user knows where the CLI downloads files, they could delete the partial download for another attempt.

## Environment

* Windows 10 Pro 22H2: 19045.1949
* Visual Studio 2022 Enterprise: 17.3.1
* VS 2022 Tools for Meadow [Preview]: v0.19.6
* Meadow.CLI: v0.19.4

adrianstevens, 2023-01-09:
@patridge great call - I'll investigate - should be an easy fix

adrianstevens, 2023-08-29:
This is fixed - closing issue

closed 2023-08-29 by adrianstevens