WildernessLabs/Meadow_Issues #225: Awaited wi-fi connection doesn't have an IP address yet

patridge, 2022-09-08:
When connecting to a wi-fi network, the awaited statement returns before the connection is actually done and an IP address is available.

## Steps

1. Create a Meadow app that attempts to connect to an existing network with accurate credentials and output the IP address after the connection is awaited.

You'll get `0.0.0.0` instead of the actual IP address, which you can get later 

### Example Code

```csharp
string wifiName = "<some-ssid>";
string wifiPassword = "<ssid-password>"; // WARNING: Not a good practice to hard-code directly into source.
IWiFiNetworkAdapter wifi;
wifi = Device.NetworkAdapters.Primary<IWiFiNetworkAdapter>();
Console.WriteLine($"Connecting to WiFi Network {wifiName}");
try
{
    var connectionResult = await wifi.Connect(wifiName, wifiPassword, TimeSpan.FromSeconds(45));
    if (connectionResult.ConnectionStatus != ConnectionStatus.Success)
    {
        throw new Exception($"Cannot connect to network: {connectionResult.ConnectionStatus}");
    }
    else
    {
        Console.WriteLine($"Connected to Wi-Fi");
        IPAddress zeroIp = IPAddress.Parse("0.0.0.0");
        var timer = Stopwatch.StartNew();
        while (wifi.IpAddress.Equals(zeroIp))
        {
            Thread.Sleep(0);
        }
        timer.Stop();
        Console.WriteLine($"Waited: {timer.ElapsedMilliseconds}ms");
        Console.WriteLine($"IP address: {wifi.IpAddress}");
    }
}
catch (Exception ex)
{
    Console.WriteLine($"Failed to connect to Wi-Fi: {ex}");
    throw;
}
```

## Expected

In the example code, I'd expect to have an IP address immediately after the `await wifi.Connect(...)` line returns.

## Actual

In the example code, assuming you can connect to the given network, you will get output like this.

```console
Meadow StdOut: Connecting to WiFi Network <some-ssid>
Meadow StdOut: Connected to Wi-Fi
Meadow StdOut: IP address: 0.0.0.0
Meadow StdOut: Waited: 160ms
Meadow StdOut: IP address: 192.168.0.168
```

## Workaround

As in the example above, you could wait through a while loop until the IP address resolves to something other than `0.0.0.0`. This usually happens between 50-200ms.

```csharp
// HACK: Work around IP not being assigned yet.
IPAddress zeroIp = IPAddress.Parse("0.0.0.0");
while (wifi.IpAddress.Equals(zeroIp))
{
    Thread.Sleep(50); // Set to whatever value you want.
}
```

Alternatively, once the adapter's `NetworkConnected` event is triggered the second time, once for `0.0.0.0` and once again after that (different issue, see #220), the IP address will have been established.

## Environment

* Windows 10 Pro 22H2: 19045.1949
* Visual Studio 2022 Enterprise: 17.3.1
* VS 2022 Tools for Meadow [Preview]: v0.19.6
* Meadow.CLI: v0.19.4
* Meadow OS: v.0.6.7.11
* Meadow projects latest from `develop` as of 07 September 2022

ctacke, 2022-09-09:
PR #199 fixes this

patridge, 2022-10-27:
This appears to be live in the latest code. Hoping that means it's out for all folks on RC1.

patridge, 2022-10-27:
Closing as resolved.

closed 2022-10-27 by patridge