WildernessLabs/Meadow_Issues #229: build-info.json is invalid JSON

ctacke, 2022-09-11:
Some time after the 0.6.4.0 release, the firmware `build-info.json` file started containing invalid json. The error appears the same in all: an invalid comma at the end of line 3.

Here's an example from 0.6.6.4. Not the end of the line beginning with `"meadow":`

```
{
  "git": {
    "meadow": [ "c46c1a44d78f5affc6107555d7a2c84af669a6f7", "" ],
  },
  "build-date": "2022-07-21 11:14:16",
  "build-hash": "a666afb75ade577e2f0b3707e9a42c8264f187acb0f4c25817935541c35fc7dd"
}
```

alexischr, 2022-09-13:
I probably introduced this one when removing the submodule hashes we used to have. I did think JSON allowed trailing commas but I was wrong.

NevynUK, 2022-10-27:
Fixed.

closed 2022-10-27 by NevynUK