WildernessLabs/Meadow_Issues #231: [v0.6.7.18] Socket error when making HTTP request after handled SSL validation exception

patridge, 2022-09-19:
[Note: This is for a pre-release version of Meadow OS: v0.6.7.18.]

I have a simple app that requests several websites with bad SSL configurations, such as self-signed certificate or an expired certificate. Even though I catch these exceptions and allow the app to move on, the next request will fail with an exception. (This happens regardless of whether the next request targets a bad SSL configuration or not.)

A sample Meadow app is below.

## Expected

Should be able to catch and handle, or ignore, an SSL exception and then make any follow-up HTTP requests needed.

## Actual

The next HTTP request after an SSL validation exception will fail with some sort of "duplicate socket" error, regardless of the SSL configuration of the destination URL.

```console
[2022-09-18 21:14:04] Meadow StdOut: Requesting https://self-signed.badssl.com/ - 01/01/2000 01:38:17
[2022-09-18 21:14:19] Meadow StdOut: Request exception: The SSL connection could not be established, see inner exception.
Requesting https://postman-echo.com/get?foo1=bar1&foo2=bar2 - 01/01/2000 01:38:33
[2022-09-18 21:14:19] Meadow StdOut: mono_fdhandle_insert: duplicate Socket fd 16
```

## Sample code

```csharp
IWiFiNetworkAdapter wifi;

public override Task Initialize()
{
    Console.WriteLine("Initialize...");
    wifi = Device.NetworkAdapters.Primary<IWiFiNetworkAdapter>();
    return base.Initialize();
}

public override async Task Run()
{
    Console.WriteLine("Run...");
    Console.WriteLine($"Connecting to WiFi Network {Secrets.WIFI_NAME}");
    try
    {
        var connectionResult = await wifi.Connect(Secrets.WIFI_NAME, Secrets.WIFI_PASSWORD, TimeSpan.FromSeconds(45));
        if (connectionResult.ConnectionStatus != ConnectionStatus.Success)
        {
            throw new Exception($"Cannot connect to network: {connectionResult.ConnectionStatus}");
        }

        Console.WriteLine($"Connected to Wi-Fi");
        // HACK: Work around IP address acquisition delay (WildernessLabs/Meadow_Issues#225).
        var timer = Stopwatch.StartNew();
        IPAddress zeroIp = IPAddress.Parse("0.0.0.0");
        while (wifi.IpAddress.Equals(zeroIp))
        {
            Thread.Sleep(50);
        }
        timer.Stop();
        Console.WriteLine($"IP address: {wifi.IpAddress}");

        // Test various bad SSL requests.
        await GetWebPageViaHttpClient("https://self-signed.badssl.com/");
        await GetWebPageViaHttpClient("https://postman-echo.com/get?foo1=bar1&foo2=bar2");
        await GetWebPageViaHttpClient("https://expired.badssl.com/");
        await GetWebPageViaHttpClient("https://hsts.badssl.com/");
        await GetWebPageViaHttpClient("https://rc4.badssl.com/");
    }
    catch (Exception ex)
    {
        Console.WriteLine($"Failed to connect to Wi-Fi: {ex}");
        throw;
    }

    await base.Run();
}

public async Task GetWebPageViaHttpClient(string uri)
{
    Console.WriteLine($"Requesting {uri} - {DateTime.Now}");
    using (var client = new HttpClient())
    {
        try
        {
            HttpResponseMessage response = await client.GetAsync(uri);
            response.EnsureSuccessStatusCode();
            string responseBody = await response.Content.ReadAsStringAsync();
            Console.WriteLine($"Success [length: {responseBody.Length}]");
        }
        catch (TaskCanceledException cancelledEx)
        {
            Console.WriteLine($"Request timed out: {cancelledEx.Message}");
        }
        catch (Exception e)
        {
            Console.WriteLine($"Request exception: {e.Message}");
        }
    }
}
```

## Environment

* Windows 10 Pro 22H2: 19045.2006
* Visual Studio 2022 Enterprise: 17.3.1
* VS 2022 Tools for Meadow [Preview]: v0.19.6 (fix for Meadow connection issue on deploy)
* Meadow.CLI: v0.19.8
* Meadow OS: v.0.6.7.18
* Meadow projects latest from `develop` as of 18 September 2022

patridge, 2022-09-29:
In case it helps, I just confirmed this is still happening for Meadow OS v.0.6.7.31.

alexischr, 2022-10-11:
The issue has been reproduced, patch incoming

patridge, 2022-10-27:
I'm still seeing this issue in the current latest source-built libraries. Here, it made a regular request fine. Then a bad-SSL request with an exception. Then the next bad request failed on the socket error. (I believe it would have failed on a good request at that point as well, but haven't confirmed.)

```console
[2022-10-26 19:18:02] Meadow StdOut: Regular request
[2022-10-26 19:18:03] Meadow StdOut: Requesting https://postman-echo.com/get?foo1=bar1&foo2=bar2 - 10/27/2022 01:18:02
[2022-10-26 19:18:28] Meadow StdOut: Success [length: 247]
[2022-10-26 19:18:28] Meadow StdOut: Regular request: self-signed
Requesting https://self-signed.badssl.com/ - 10/27/2022 01:18:28
[2022-10-26 19:18:30] Meadow StdOut: Request exception: System.Net.Http.HttpRequestException: The SSL connection could not be established, see inner exception. ---> System.IO.IOException: TLS initialization or handshake failed
  at Mono.MbedTls.MbedTlsContext..ctor (Mono.Net.Security.MobileAuthenticatedStream mas_stream, Mono.Net.Security.MonoSslAuthenticationOptions options, System.Runtime.InteropServices.SafeHandle[2022-10-26 19:18:30] Meadow StdOut: socket_handle, System.Net.Sockets.NetworkStream network_stream) [0x0008d] in <fd022122478e4c3293eb61a1e7cbdebe>:0 
  at Mono.MbedTls.MbedTlsStream.CreateContext (Mono.Net.Security.MonoSslAuthenticationOptions options) [0x00012] in <fd022122478e4c3293eb61a1e7cbdebe>:0 
  at Mono.Net.Security.MobileAuthenticatedStream.ProcessAuthentication (System.Boolean runSynchronously, Mono.Net.[2022-10-26 19:18:30] Meadow StdOut: ecurity.MonoSslAuthenticationOptions options, System.Threading.CancellationToken cancellationToken) [0x0014c] in <fd022122478e4c3293eb61a1e7cbdebe>:0 
  at System.Net.Http.ConnectHelper.EstablishSslConnectionAsyncCore (System.IO.Stream stream, System.Net.Security.SslClientAuthenticationOptions sslOptions, System.Threading.CancellationToken cancellationToken) [0x000c3] in <ee2b481e[2022-10-26 19:18:30] Meadow StdOut: f0c4d11b88b950fb31d5ef5>:0 
   --- End of inner exception stack trace ---
  at System.Net.Http.ConnectHelper.EstablishSslConnectionAsyncCore (System.IO.Stream stream, System.Net.Security.SslClientAuthenticationOptions sslOptions, System.Threading.CancellationToken cancellationToken) [0x00102] in <ee2b481e4f0c4d11b88b950fb31d5ef5>:0 
  at System.Net.Http.HttpConnectionPool.CreateCo[2022-10-26 19:18:30] Meadow StdOut: nectionAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) [0x00322] in <ee2b481e4f0c4d11b88b950fb31d5ef5>:0 
  at System.Net.Http.HttpConnectionPool.WaitForCreatedConnectionAsync (System.Threading.Tasks.ValueTask`1[TResult] creationTask) [0x000a2] in <ee2b481e4f0c4d11b88b950fb31d5ef5>:0 
  at System.Net.Http.HttpConnectionPool.[2022-10-26 19:18:30] Meadow StdOut: endWithRetryAsync (System.Net.Http.HttpRequestMessage request, System.Boolean doRequestAuth, System.Threading.CancellationToken cancellationToken) [0x00089] in <ee2b481e4f0c4d11b88b950fb31d5ef5>:0 
  at System.Net.Http.RedirectHandler.SendAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) [0x00094] in <ee2b481e4f0c4d11b88b950fb[2022-10-26 19:18:30] Meadow StdOut: 1d5ef5>:0 
  at System.Net.Http.HttpClient.FinishSendAsyncBuffered (System.Threading.Tasks.Task`1[TResult] sendTask, System.Net.Http.HttpRequestMessage request, System.Threading.CancellationTokenSource cts, System.Boolean disposeCts) [0x0016c] in <ee2b481e4f0c4d11b88b950fb31d5ef5>:0 
  at MeadowApplication202208100924vs2022win.NetworkHelpers.GetWebPageViaHttpClient (System.String [2022-10-26 19:18:30] Meadow StdOut: ri) [0x000bc] in <c2b3419981664a01aee656947ea1234b>:0 
	System.IO.IOException: TLS initialization or handshake failed
  at Mono.MbedTls.MbedTlsContext..ctor (Mono.Net.Security.MobileAuthenticatedStream mas_stream, Mono.Net.Security.MonoSslAuthenticationOptions options, System.Runtime.InteropServices.SafeHandle socket_handle, System.Net.Sockets.NetworkStream network_stream) [0x0008[2022-10-26 19:18:30] Meadow StdOut: ] in <fd022122478e4c3293eb61a1e7cbdebe>:0 
  at Mono.MbedTls.MbedTlsStream.CreateContext (Mono.Net.Security.MonoSslAuthenticationOptions options) [0x00012] in <fd022122478e4c3293eb61a1e7cbdebe>:0 
  at Mono.Net.Security.MobileAuthenticatedStream.ProcessAuthentication (System.Boolean runSynchronously, Mono.Net.Security.MonoSslAuthenticationOptions options, System.Threading.Cancella[2022-10-26 19:18:31] Meadow StdOut: ionToken cancellationToken) [0x0014c] in <fd022122478e4c3293eb61a1e7cbdebe>:0 
  at System.Net.Http.ConnectHelper.EstablishSslConnectionAsyncCore (System.IO.Stream stream, System.Net.Security.SslClientAuthenticationOptions sslOptions, System.Threading.CancellationToken cancellationToken) [0x000c3] in <ee2b481e4f0c4d11b88b950fb31d5ef5>:0 
Regular request: expired
Requesting https:/[2022-10-26 19:18:31] Meadow StdOut: expired.badssl.com/ - 10/27/2022 01:18:30
mono_fdhandle_insert: duplicate Socket fd 16
```

duduita, 2024-01-25:
After a quick test using v1.8.0.0, this issue looks fixed, since there's no duplicate socket descriptor error anymore after an SSL exception:

```
Meadow StdOut: Requesting https://self-signed.badssl.com/ - 01/16/2024 20:12:06
Meadow StdOut: Handshake failed
Meadow StdOut: Request exception: The SSL connection could not be established, see inner exception.
Meadow StdOut: Requesting https://postman-echo.com/get?foo1=bar1&foo2=bar2 - 01/16/2024 20:12:24
Meadow StdOut: Success [length: 302]
Meadow StdOut: Requesting https://expired.badssl.com/ - 01/16/2024 20:12:36
Meadow StdOut: Handshake failed
Meadow StdOut: Request exception: The SSL connection could not be established, see inner exception.
Meadow StdOut: Requesting https://hsts.badssl.com/ - 01/16/2024 20:12:38
Meadow StdOut: Handshake failed
Meadow StdOut: Request exception: The SSL connection could not be established, see inner exception.
Meadow StdOut: Requesting https://rc4.badssl.com/ - 01/16/2024 20:12:39
Meadow StdOut: Handshake failed
Meadow StdOut: Request exception: The SSL connection could not be established, see inner exception.
```

closed 2024-03-19 by adrianstevens